OrcaSlicer/OrcaSlicer #6260: Modal dialogs open on a different desktop on Mac OS

stefansjs, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Printer

Bambu Lab A1 mini

### How to reproduce

Modal dialogs open on a different "desktop" than the OrcaSlicer window. This happens any time the printer wants to prompt the user, for example errors that stop the printer like filament runout, or intentional pauses that request user interaction like M400 U1.

1. Configure a multimaterial, single-extruder printer configuration
2. Enable "Manual Filament Change"
3. Replace the "Change filament G-Code" under "Machine G-Code" with: "M400 U1; Pause and wait for the user"
4. slice a model
5. add a filament change at some layer
6. print until the filament change appears

### Actual results

A modal appears on another desktop. When I click on the OrcaSlicer window, it throws me to another desktop with the modal. When I dismiss the modal, it doesn't return me to the OrcaSlicer desktop

### Expected results

The modal should appear in the same desktop space as the OrcaSlicer window.

### Project file & Debug log uploads

[modal-dialog-bug.3mf.zip](https://github.com/user-attachments/files/16395959/modal-dialog-bug.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

richard-scott, 2024-10-14:
Also, modal windows appear behind OrcaSlicer (and any other windows), making them really hard to find.

closed 2025-01-22 by github-actions[bot]

Rubikoid, 2026-10-17:
Still a problem in 2.4.2