OrcaSlicer/OrcaSlicer #6261: Frequent crashes when switching tabs in preset windows

kingr0man, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Manjaro (Arch) [kernel 6.9.9] + KDE plasma [v 6.0.5]

### Additional system information

OS: Manjaro Linux x86_64
Host: 82V2 (Slim 7 ProX 14ARH7)
Kernel: Linux 6.9.9-1-MANJARO
Display (CSO1410): 3072x1920 @ 120Hz (as 1536x960) [Built-in]
DE: KDE Plasma 6.0.5
WM: KWin (Wayland)
CPU: AMD Ryzen 9 6900HS Creator Edition (16) @ 4.93 GHz
GPU 1: AMD Radeon 680M [Integrated]
GPU 2: NVIDIA GeForce RTX 3050 Mobile [Discrete]
Memory: 27.10 GiB

### Printer

any

### How to reproduce

1. Click the 'Edit' button on any printer or filament preset
2. Click on another tab in the preset window (ie try to switch from 'Basic information' to 'Machine G-code' in a printer preset)
3. Crash 

### Actual results

Program freezes for a second, then closes itself

### Expected results

Tab switches

### Project file & Debug log uploads

Here's my full log folder containing several crash logs, mostly if not all from this issue. 
[log.zip](https://github.com/user-attachments/files/16396171/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

discip, 2024-07-27:
Same behavior here!
Also on Linux (Manjaro).

@SoftFever
@Noisyfox
Please have a look, this makes it unusable.

mttr402, 2024-07-28:
Having the same issue in Linux mint 21.3 Mate with all Orca versions of V2.x.  One thing Orca does not like - (dash) in file names as it reports as errors. I tested by renaming the file and it is not flag in log as an error. The fault also occur with a new project started and with no objects added.  
[error]	2024-07-28 12:41:16.777044[Thread 0x00007f41e74db380]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-07-28 12:41:16.804953[Thread 0x00007f41e74db380]:Unable to open the file/media/3D.objects/Collar.Insert.For.Dremel/Collar.Insert - Collar.Full.3mf
[error]	2024-07-28 12:41:16.804967[Thread 0x00007f41623fd640]:Unable to open the file/home/Downloads/Plier.Plate.step
[error]	2024-07-28 12:41:16.804989[Thread 0x00007f4160bfa640]:Unable to open the file/media/3D.objects/Collar.Insert.For.Dremel/Collar.Insert - Collar.Full.step
[error]	2024-07-28 12:41:16.805029[Thread 0x00007f41633ff640]:Unable to open the file/home/Downloads/LG.Pump.Damper - Full.Part(1).step
[error]	2024-07-28 12:41:16.805031[Thread 0x00007f4160bfa640]:Unable to open the file/home/Downloads/C9.Standard.Bulb.Extractor - Part.Main.Body.step
[error]	2024-07-28 12:41:16.805116[Thread 0x00007f4160bfa640]:Unable to open the file/home/Downloads/Throotle.control.arm.step
[error]	2024-07-28 12:41:16.805108[Thread 0x00007f41692fb640]:Unable to open the file/media/3D.objects/Josie/shot-blocker.STL
[error]	2024-07-28 12:41:16.805174[Thread 0x00007f41e74db380]:Unable to open the file/home/Downloads/Plier.Plate.V3.step
[error]	2024-07-28 12:41:16.805208[Thread 0x00007f4161bfc640]:Unable to open the file/home/Downloads/Hose.Reducer.v1 - Hose.Reducer.v1.full.step
[error]	2024-07-28 12:41:16.805221[Thread 0x00007f41e74db380]:Unable to open the file/home/Downloads/Plier.Plate.3mf
[warning]	2024-07-28 12:41:17.566850[Thread 0x00007f41e74db380]:12:41:17 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-28 12:41:17.570003[Thread 0x00007f41e74db380]:12:41:17 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-28 12:41:17.578320[Thread 0x00007f41e74db380]:12:41:17 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-28 12:41:17.585615[Thread 0x00007f41e74db380]:12:41:17 PM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-07-28 12:41:19.862480[Thread 0x00007f41e74db380]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-07-28 12:41:19.908688[Thread 0x00007f41e74db380]:get_version, get_version not supported,return 00.00.00.00!
[error]	2024-07-28 12:41:29.659737[Thread 0x00007f41e74db380]:calc_exclude_triangles:Unable to create exclude triangles

adrianbear, 2024-08-09:
This occurs when changing a value in a field (for me, particularly a field in one of the extruder tabs) and then changing tabs (to another extruder tab). If you change from the edited field, to another field within the same tab before changing tabs, the error does not seem to occur.

Noisyfox, 2024-08-10:
> This occurs when changing a value in a field (for me, particularly a field in one of the extruder tabs) and then changing tabs (to another extruder tab). If you change from the edited field, to another field within the same tab before changing tabs, the error does not seem to occur.

Nice, I managed to reproduce the issue in my VM using this method. Will look into this problem soon.

Jakeo-93, 2024-08-10:
getting this to without changing values as well. Arch based os as well ill just say it is unstable well in the preset tabs.

Noisyfox, 2024-08-11:
> getting this to without changing values as well. Arch based os as well ill just say it is unstable well in the preset tabs.

Yeah I noticed that too.

Noisyfox, 2024-08-12:
Please give the PR https://github.com/SoftFever/OrcaSlicer/pull/6437 a try and see if it solves the problem.

discip, 2024-08-12:
@Noisyfox
# Thank you very much Sir! 😃👍
This solved the issue!

But unfortunately there is more:
This orange symbol is always there except on the **`Multimaterial tab`**, even if nothing was changed:

[image]

After changing any parameter, another symbol appears underneath:

[image]

If you are lucky enough to be able to klick it (the one underneath) everything works fine, but if you klick the one on top of it, especially after switching tabs, it crashes and closes Orca.

To reproduce:
1. open Orca
2. enter **`Printer settings`**
3. click the **`Machine G-code tab`**
4. click the orange symbol in the top left corner

Noisyfox, 2024-08-13:
Hmmmmm seems to be another Linux only issue. Will take a look soon.

Noisyfox, 2024-08-13:
> @Noisyfox
> 
> # Thank you very much Sir! 😃👍
> This solved the issue!
> 
> But unfortunately there is more: This orange symbol is always there except on the **`Multimaterial tab`**, even if nothing was changed:
> 
> ![image](https://private-user-images.githubusercontent.com/53649486/357148837-e1ff51d0-e26f-4311-880b-d68bb16e0fc8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1MTA4MjksIm5iZiI6MTcyMzUxMDUyOSwicGF0aCI6Ii81MzY0OTQ4Ni8zNTcxNDg4MzctZTFmZjUxZDAtZTI2Zi00MzExLTg4MGItZDY4YmIxNmUwZmM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEzVDAwNTUyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmMWY0NzRjYTBhMTVhMzNhOTdkMGM0N2ZmYTRjOWYyYzEwZjhmMTE5MjNmNzZjOWNjNTkxNjdkZGQwM2VjNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DMxDt2d1NjfCYnG_rKNF0BsCsBioIHP1403f_YmKZAQ)
> 
> After changing any parameter, another symbol appears underneath:
> 
> ![image](https://private-user-images.githubusercontent.com/53649486/357148935-2e181bdf-4db8-4f9b-ad93-0646041a603e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1MTA4MjksIm5iZiI6MTcyMzUxMDUyOSwicGF0aCI6Ii81MzY0OTQ4Ni8zNTcxNDg5MzUtMmUxODFiZGYtNGRiOC00ZjliLWFkOTMtMDY0NjA0MWE2MDNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEzVDAwNTUyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjMThiNzY4MzA2ZTc1ODEyYjViMzdlY2E3MWFhNDk1MDM3ODk0ZDA5NjllZWRiOTZjYjE1ODQ2MmE0MmZmMzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Z6Gd9K_HHIqNwIF1KnqSf_9XmW1NeA-1Yy_uJv3ie-w)
> 
> If you are lucky enough to be able to klick it (the one underneath) everything works fine, but if you klick the one on top of it, especially after switching tabs, it crashes and closes Orca.
> 
> To reproduce:
> 
> 1. open Orca
> 2. enter **`Printer settings`**
> 3. click the **`Machine G-code tab`**
> 4. click the orange symbol in the top left corner
> 
> [log1.txt](https://github.com/user-attachments/files/16590658/log1.txt) [log2.txt](https://github.com/user-attachments/files/16590659/log2.txt) [log3.txt](https://github.com/user-attachments/files/16590660/log3.txt)

Try the latest PR build. I don't have access to my Linux VM at the moment so I could not test this myself, but hopefully it's fixed now.

discip, 2024-08-13:
@Noisyfox
# Thank you very much! 😃👍
That did indeed solve the issue!

PR #6437 (2024-08-18): Fix crash on Linux when switching between tabs in printer config
This fixes #6261.
closed 2024-08-18 by SoftFever