OrcaSlicer/OrcaSlicer #6263: Error: in GCode file line 164: G2: outside machine limits Error: M25: Machine is halted

moth4017, 2024-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

windows 64 pro

### Printer

Valkyrie HT

### How to reproduce

slice object  when  start print , error because it think its outside the limits 

### Actual results

Cancelled printing file 0:/gcodes/295x295_PLA_5m33s.gcode, print time was 0h 3m
Error: in GCode file line 164: G2: outside machine limits
Error: M25: Machine is halted

### Expected results

no error , added go code to report and pictures of the printer set up
[image]
[image]

### Project file & Debug log uploads

[295x295_PLA_5m33s.zip](https://github.com/user-attachments/files/16397820/295x295_PLA_5m33s.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

buzzhuzz, 2024-07-27:
I've looked into the gcode and see that G2 reported by machine has center of the arc outside of build plate:
[image]

Seems like machine FW does not like G2/G3 centers be outside of the build plate. We probably need to add special safety check for this as a gcode post processor to warn user about potential problem.

You could try to disable "arc fitting" in our print profile which may help you print this exact model.

Also, I curious to see 3mf file with the model as I don't see obvious reason why G2 arcs are there.

moth4017, 2024-07-27:
Thank you for the update , and a very clear  explanation of the issue , I have disabled the arc  fitting  and retrying the print today.

Re G2/G3  in RRF , have asked a question to the DEV team about this.

Mark

g2-error-explain.png (view on web)<https://github.com/user-attachments/assets/f26a4cfb-03fe-4939-a74c-a8b904a7d484>

closed 2024-11-03 by github-actions[bot]