OrcaSlicer/OrcaSlicer #6264: "Bridges do not overlap perimeters which result in failed print"

vadixidav, 2024-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

This is a known bug. See https://github.com/prusa3d/PrusaSlicer/issues/9999. It was fixed upstream in PrusaSlicer. It has not been merged to Orca.

Slice any model which has a thin section with more wall loops enabled than can be printed and a bridge subsequently next to that spot. Other situations can trigger this as well.

### Actual results

Again, see https://github.com/prusa3d/PrusaSlicer/issues/9999 or https://github.com/prusa3d/PrusaSlicer/issues/11150.

Here is an example of the outcome on a model I am printing:

[image]

Note that the bridge is not anchored to the print at all and simply prints in mid-air. This should be well understood already by most people since this happens for most functional prints (>50%)  if they contain a bridge and when Orca Slicer is used with 3 or more walls. For people who use 2 or 1 wall, this may not occur as often as the number of walls is not sufficiently high to prevent the bridge. Most authors worked around this issue by modifying their model so that Prusa and Orca slicer would print them correctly with 2 walls at most, but many authors did not care to support the 3 wall case. Now this is fixed for any number of wall loops.

### Expected results

The bridge should be anchored. See https://github.com/prusa3d/PrusaSlicer/issues/9999. Fixed in Prusa slicer 2.7.3.

### Project file & Debug log uploads

The project I am currently working on has 11 parts in it and it would only be confusing to upload. Please see the Prusa slicer issue for details <https://github.com/prusa3d/PrusaSlicer/issues/9999>. There are also other issues open on Orca Slicer which may be caused by this issue, but are other bridging-related issues which were fixed in Prusa Slicer, but I don't believe they have been properly connected to this underlying issue yet.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I think contributors are already aware of this ongoing related work due to the mention of the PrusaSlicer issue in the discussion, but also see https://github.com/SoftFever/OrcaSlicer/pull/4651. This issue is just to formally note that the bridge fixes should be merged into Orca Slicer.

digital0785, 2024-08-09:
yess! there should be at least a bridge wall overlap at minimum.. as well as a slow bridge layer count as someone else has suggested so it doesn't go from a slow layer then back to full speed as the combo of these two dramatically reduces quality of bridging

jamincollins, 2024-08-20:
I've tested slicing a model I'm trying to print several different ways in Orca Slicer and find that no matter the number of perimeters (walls) I select or the orientation of the bridge, Orca attempts to create bridges that are either not anchored or attempt to run the longest path on the model.

Longest path:
[image]
[image]

Unanchored (mid-air):
[image]
[image]

jamincollins, 2024-08-20:
While the same model with similar settings slices much more sanely in Prusa Slicer:
[image]

adolfotregosa, 2024-10-27:
Same problem over here. Reducing walls loops to 2 helps.

4 wall loops:
[image]

2 wall loops:
[image]

digital0785, 2025-01-27:
Please integrate this from prusa it'll be an overall improvement

Dedoque, 2025-03-22:
+ for it! Bridging is a pain in orca!

digital0785, 2025-03-22:
> plus for it! Bridging is a pain in orca!

this has been fixed in 2.3 i believe as well as multiple bridging layers

Dedoque, 2025-03-23:
[image] doesn`t seems like a fix

digital0785, 2025-03-23:
> [image] doesn`t seems like a fix

you're not showing an external bridge layer which is what this is referring too the only reason an internal bridge would have an issue is because theres too long a span without infill which is not a slicer issue. notice how the SS i posted it deviates from the wall path to give the bridge an anchor   

[image]

vadixidav, 2025-03-25:
Yes, I didn't explicitly mention external bridges, but this was specifically in relation to external bridges. The issue should not be closed though. The bot merely closed it because it went stale. It is not fixed as far as I am aware.

Bata3, 2025-03-30:
Enabling Ensure vertical shell thickness helped me. 
When it was turned off, there was only overhang on one perimeter.

[image]

off:
[image]

on:

[image]

The difference in quality was huge.

[image]

digital0785, 2025-03-31:
i have it on moderate that could be why i didn't notice :)

vadixidav, 2025-03-31:
> Enabling Ensure vertical shell thickness helped me. 
> When it was turned off, there was only overhang on one perimeter.
> 
> [image]
> 
> off:
> [image]
> 
> all:
> [image]
> 
> The difference in quality was huge.
> 
> [image]

Did you also try changing the number of walls? I believe that if you increase the number of walls that you would still get a non-overlapping bridge. That being said, it is good to see that in some situations the bridging behavior can be worked around.

Bata3, 2025-03-31:
> 
> Skúsili ste tiež zmeniť počet stien? Verím, že ak zvýšite počet stien, stále by ste dostali neprekrývajúci sa most. Ako už bolo povedané, je dobré vidieť, že v niektorých situáciách je možné premosťovacie správanie obísť.

It doesn't matter how many perimeters there are, the bridging overlap will always be correct.

[image]

[image]

GroovybandLive, 2025-04-03:
The bridge feature in Orca Slicer is still completely broken (2.3.0):

[image]

In practice the slicer is unusable. This issue has been there for years now!! Instead of adding questionable features, the developers should solve this fundamental bug. Probably they use some other slicer for their daily prints!!!

yeahdino, 2025-04-07:
I think what is happening is that Orca doesn't want to overlap areas that are just perimeters. A lame workaround is to use the height range modifier to decrease the perimeters in the layer having issues. This issue also exists in Bambu Slicer.

[image]

kniftar, 2025-04-09:
I've been struggling with this for a week. I'm running 2.3.0.  I tried the latest nightly as of today because there was a commit entry that claimed to fix this. Well it did - until I imported my profile then it returned.  That prompted me to check every setting again and I finally determined setting "Bridge counterbore holes" to "none" fixes it.  So I hope that might be a clue that could get this fixed.

jkortus, 2025-06-11:
> I've been struggling with this for a week. I'm running 2.3.0. I tried the latest nightly as of today because there was a commit entry that claimed to fix this. Well it did - until I imported my [...]

I had to change the settings from "none" to anything else and it fixed my print. I scaled my model down beyond the treshold discussed here with the walls and the bridges started to collapse. The threshold in the model for me is on 0.4mm nozzle around 0.8mm thick walls, above it works, below it collapses.

closed 2025-10-20 by github-actions[bot]