OrcaSlicer/OrcaSlicer #6266: Avoid crossing walls feature removes some retraction wipes

bernarden, 2024-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality Ender 5 S1

### How to reproduce

1. I used the following default profiles `Creality Ender-5 S1 0.4 nozzle`, `Creality Generic PLA` and `0.20mm Standard @Creality Ender5S1` but pretty sure it can be reproduced with other profiles.
2. Generate `Retraction test` model (start: 0mm, end: 2mm, step: 0.1mm/mm).
3. Change retraction wipe distance on printer setting to 10mm so it's very obvious when the bug occurs.
[image]

5. No overrides on filament level.
6. Tick `Avoid crossing walls`
7. Optionally set `Max detour length` to some value other than 0. (No impact changing this value)
8. Slice the model and enable `Travel` line type in preview.

### Actual results

On **some** layers one of the sides does not have a retraction wipe gcode generated. The results are very sporadic. When looking at any 10 layers at the time this issue could affect 0-2 lines but sometimes more like 7-8 lines. In any case this was leading to constant stringing no matter what retraction setting changes I did. 
[2024-07-27.webm](https://github.com/user-attachments/assets/378962da-6e82-4c36-8662-def99a778557)

Also, you can observe on the side where the wipe gcode is generated the nozzle still crosses the walls anyway, but this perhaps is related more to #5217.

### Expected results

Retraction wipe gcode is generated for both sides of the model. 

### Project file & Debug log uploads

[Retraction test.zip](https://github.com/user-attachments/files/16399006/Retraction.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I've tested a few previous versions to see if it ever worked and got the following results:
1.6.6 - works
1.8.1 - works
1.9.0  alpha - does not work
1.9.0 - does not work
1.9.1 - does not work
2.0.0 - does not work
2.1.1 - does not work

discip, 2024-07-27:
# Thank you figuring this out. 👍🏻
I was often wondering why some of my prints had the issue you described.

discip, 2024-07-27:
# Thank you figuring this out. 👍🏻
I was often wondering why some of my prints had the issue you described.

@vovodroid
@igiannakas
Could you please have a look?

vovodroid, 2024-07-28:
Actually I don't use either of these options...

discip, 2024-08-20:
@vovodroid
This helps quite a bit when printing materials that are prone to stringing (e.g. PETG).

Any idea how **`Avoid crossing walls`** affects retraction wipes?
<br>
**`Avoid crossing walls`** disabled:

[image]
<br>
**`Avoid crossing walls`** enabled:

[image]

igiannakas, 2024-08-20:
PR raised: https://github.com/SoftFever/OrcaSlicer/pull/6518

If you fancy, please give it a test.

igiannakas, 2024-08-20:
The odd part is that this whole code segment plus all the code that executes the avoid crossing walls functionality is basically unchanged since the orca first fork. So seeing this behave differently between versions is unexplained but may need to be examined further.

I would expect the model shown above to show similar behaviour across all versions. It is a good test, if anyone fancies checking against 1.8.1.

bernarden, 2024-08-21:
@igiannakas You are correct that the issue is present in 1.8.1. I haven't noticed it since only a few layers were missing the wipe operation. Since that version the issue kept getting worse though.
[image]
[image]
[image]
[image]
[image]
[image]

igiannakas, 2024-08-21:
Hm this makes no sense as the immediate code has been unchanged - I even compared it against the curent prusa slicer code and it’s functionally identical 🤔 

In any case the fix above should sort this once and for all as it enforces a wipe move when a retraction is being done, just like what the slicer does when avoid crossing walls is disabled.

I’d appreciate if you could test the build in the PR above and offer feedback please.

closed 2024-08-27 by SoftFever