OrcaSlicer/OrcaSlicer #6268: Incorrect preview color in "Send print job to..." dialog

TinyMito, 2024-07-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

14

### Printer

P1S + 2x AMS

### How to reproduce

1. Model with two or more color.
2. Objects are individual coloured, not painted.
3. Send print job to see the preview often enough to confused me.

### Actual results

Color flipped.

[image]

### Expected results

Should display the correct color as it is in the Preview tab.

### Project file & Debug log uploads

[XTouch+Total+Screen+Replacement+(Front+and+Back).3mf.zip](https://github.com/user-attachments/files/16400961/XTouch%2BTotal%2BScreen%2BReplacement%2B.Front%2Band%2BBack.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Arakon, 2024-07-28:
Same as #5765 I believe.

tlhintoq, 2024-07-29:
And same as https://github.com/SoftFever/OrcaSlicer/issues/6203

closed 2024-08-05 by TinyMito