OrcaSlicer/OrcaSlicer #6269: Add Kobra 3 to printer profiles

deliciousCodeSandwich, 2024-07-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I just downloaded Orca slicer and I was planning on trying it for my new Kobra 3 Combo, but it turns out there is no printer profile for the Kobra 3.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Please add the Kobra 3 to printer profiles. I am new to 3d printing and I'm not familiar with slicers and stuff and I don't know if there is an alternative printer profile that is the 'same' as the Kobra 3, but I'm just not gonna risk it. Until then, I will use PrusaSlicer. Thanks!

exussum12, 2024-07-29:
https://wiki.anycubic.com/en/fdm-3d-printer/kobra-3-combo/orca-import-k3-parameter-steps

Has a beta profile. No network support etc added to it either.

deliciousCodeSandwich, 2024-07-31:
I tried it and I got stuck on the step where you import the K3 parameter
package, as it assumes you already have those files, and I don't.

exussum12, 2024-07-31:
They are linked at the top of the Wiki

soosp, 2024-10-02:
See: https://github.com/SoftFever/OrcaSlicer/issues/5048#issuecomment-2389296161

cochcoder, 2024-10-15:
> @cochcoder 
> 
> CAUTION: The slicer profile provided on the Wiki differs from the profile that is used in the "Anycubic Slicer Next" (Anycubic's Orca fork).
> 
> 
> 
> I opened a ticket with Anycubic but I have not gotten a reply in over a month and am not technically skilled enough to compare profile to profile as orca has no native diff ability and I am only able to text compare mentally, SO before merging into Orca someone with a better ability to compare profiles might want to download and take a gander at the profile differences!
> 
> 
> 
> 

Do you happen to know which one is better?

Hammerfest, 2024-10-15:
> Do you happen to know which one is better?

Going to have to assume (yuk) the one in the slicer as there have been changes there as well in the last 3 updates, its what makes logical sense anyway.

cochcoder, 2024-10-15:
I'm going to go with the ones from the slicer as those seem to be the best version available. However, if someone owns this printer and finds differences in print quality that support the profiles from the wiki. Please let me know!

Hammerfest, 2024-10-15:
> I'm going to go with the ones from the slicer as those seem to be the best version available. However, if someone owns this printer and finds differences in print quality that support the profiles from the wiki. Please let me know!

Appreciate it, I have 2, 1 still in the box, and on the 1 out of box I was never able to get Orca with imported profile to work properly, I saw others have, but I just was not able, so whatever they keep tweaking in their own Orca variant release, they ARE doing work on it. Still don't have more then 35hrs on it despite owning it for 3~ months, but first layer issues as with any of the prebuilts i've messed with are my get annoyed and stop messing with it phase. 

Totally love it if others with more texpertise here could chime in. Otherwise, I will for sure go through the testing I can when it makes its way to a nightly.

exussum12, 2024-10-16:
I have used all versions and the wiki download.

Personally they feel about the same to me, the difference has been the jumps in remote features (send to print, monitor print, sync settings from ace pro, discover lan printers etc)

Hammerfest, 2024-10-16:
I dont think the remote features will be coming to Orca anytime without Anycubic, but I can deal with manual push or gcode print from ASN (anycubic slicer next, aka orca but anycubic fork).

I did see at some point that the tool head dimensions changed, and seeing as I had print failures when doing print by object that's the kinda change that's not often noticed but kinda critical, a few other changes had been made with slicer vs wiki download, stuff that's behind the scenes but can wreck your day/print at the worst of times.

closed 2024-10-22 by SoftFever

Schildkroet, 2025-01-30:
> I dont think the remote features will be coming to Orca anytime without Anycubic, but I can deal with manual push or gcode print from ASN (anycubic slicer next, aka orca but anycubic fork).

What is the status of remote print now? Still not possible?

PR #7082 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3
Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes  #3332, closes #6200, closes #5048, and closes #6269
commit 09afb390 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3 (#7082)

Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes #3332, closes #6200, closes #5048, and closes #6269