OrcaSlicer/OrcaSlicer #6282: FlashForge Adventurer 3 Custom line out of printing area for every file and calibration

GabrielJanucik, 2024-07-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

22.04.4 LTS

### Additional system information

CPU: 13th Gen Intel® Core™ i9-13900H × 20
Memory: 96 GB
Graphic: NVIDIA Corporation / NVIDIA RTX 2000 Ada Generation Laptop GPU/PCIe/SSE2

### Printer

FlashForge Adventurer 3, 0.4 nozzle

### How to reproduce

1. Create new project
2. Calibration
3. Flow rate > Pass 1
4. Slice plate with default or any other presets
5. Export G-code file (That printer doesn't have network plugin in OrcaSlicer)
6. Try to print file using manufacturer app or USB stick

### Actual results

Got an error: "The printing file is out of the print range. The printer cannot print it".
In OrcaSlicer preview after slicing I can see thick straight line in start of printing. It appear always and for any file, not only in calibration. Line type shows it as "Custom".

### Expected results

No custom line or no nozzle travel outside the printing area
[image]

### Project file & Debug log uploads

[debug_Mon_Jul_29_11_08_43_579660.log.zip](https://github.com/user-attachments/files/16410514/debug_Mon_Jul_29_11_08_43_579660.log.zip)
[Flowrate Test - Pass1.zip](https://github.com/user-attachments/files/16410520/Flowrate.Test.-.Pass1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

GabrielJanucik, 2024-07-29:
Looks like I found an issue. I'm doing test print now and I will see how it goes. Eventually I will creat pull request with fix for that.

closed 2024-08-02 by SoftFever

EXE-g, 2024-12-07:
I'm confirming the problem. The problem has not been solved. Can't use G-code from OrcaSlicer because of an unknown generated line outside the table.
[image]

dbaarda, 2025-08-26:
Note this was fixed in https://github.com/SoftFever/OrcaSlicer/pull/9726 which is not yet in the official release v2.3.0, but is in the v2.3.1 aphas. However that fix only addressed the serious "out of build volume" problem and doesn't include any of the more minor things l noted like;

1. The hop-down is not over the start of the pre-extrude line.
2. There is no initial `M118` line (but this is not important?).

There are also other minor problems like;

1. It unconditionally turns on the enclosure (`M651 S255`) and extruder (`M106`) fans regardless of settings.
2. The `machine_end_gcode` turns off the enclosure fan `M652`  but not the extruder (`M107`) fan.

Note that these are mostly minor except that the Adv3 doesn't correctly understand the fan commands OrcaSlicer emits in it's generated gcode because;

1. It doesn't understand the `S<0-255>` speed argument to `M106`, and can only turn the extruder fan full on. So `M106 S0` doesn't turn off the extuder fan. Only `M107` will turn the extruder fan off.
2. It doesn't understand the `P<1-3>` fan selection argument to `M106`, and I think just ignores them. So `M106 P<2-3> S<0-255>` will not set the speed of the enclosure fan. Only `M651 S<0-255>` will set the enclosure fan speed and `M652` will turn it off.
3. It ignores any line with a comment `;` anywhere, so it ignores gcode commands with trailing comments.

This means the Adv3 ignores nearly all of the fan speed setting commands Orca generates, making setting them to the desired values in the start and end gcode more important. Personally I use these machine gcode fragments and post-processing script to modify the generated gcode to better suit the Adv3;

* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/Adv3_machine_start.gcode
* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/Adv3_machine_end.gcode
* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/Adv3_layer_change.gcode
* https://github.com/dbaarda/workspace-tools/blob/main/3DPrinting/toadv3.sh

dbaarda, 2025-09-26:
> Hi,
> 
> Thank you for the share and work done on that and I realy like my adv3 printer and I would like to better printing.
> 
> But following your work, I got some trouble to apply the post processing script. I'm on Windows / Orca Slicerr 2.3.0 . I put the file path : C:\TEMP\adv3.sh; then when I do the export , i'm getting : win32 error 193
> 
> Could you please give me some advice to apply it successfully ?

Ahh... the problem is almost certainly that my `toadv3.sh` script is a unix bash script that invokes the unix `sed` command to post-process and edit the gcode, and these tools don't exist on windows by default. It is possible to install them using things like Cygwin, and I've seen comments online that sed does exist for windows, but you might need to fiddle with things to make them work. There is also a possiblity that Unix vs DoS line endings in the text files could confuse things. If you could get a Windows sed command working, it should be fairly straight-forward to convert the bash script into a *.bat file if you needed to.

On Linux it just works.

I should also be possible to rewrite the post-processing script in Python, which may make things easier on windows. It's actually a pretty simple...

commit 74775c93 (2024-08-02): FlasForge Adventurer 3 machine start gcode fix #6282 (#6312)
commit 603e3e76 (2024-09-12): FlashForge Adventurer 3 machine start gcode second fix #6282 (#6744)