OrcaSlicer/OrcaSlicer #6296: Orca won't navigate around Exclusion Zones

robwaldhauser, 2024-07-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Custom V0.2 Toolchanger

### How to reproduce

When attempting to slice a file where items are "around the corner" of an exclusion zone or irregular bed shape

### Actual results

Orca Slicer will not attempt to calculate a path around the exclusion zone but rather try to cut straight through. What seems to happen, in the even that there can't be a straight line without cutting across the exclusion zone, it will throw an error that it is too close to the exclusion zone. If we remove the exclusion zone and set the bed size based on an stl it will simply travel outside of the bed area (which is also no good if that's where the dock is).
[image]
[image]

### Expected results

Orca should navigate to avoid exclusion zones and or custom bed sizes from stl. This is especially critical as toolchangers become more popular, where Orca will need to dodge the correctly excluded docking zones. In this case, a V0.2 has limited bed size so being about to use the space between docks for the purge tower allows far more room for the models.

### Project file & Debug log uploads

[Sheep.fail.zip](https://github.com/user-attachments/files/16428237/Sheep.fail.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

fer662, 2024-08-01:
[image]
Another example of this being an issue is some gantry designs and toolhead combinations require excluding little areas around the front corners. A travel between 2 objects located at these spots will violate the exclusion zone.

oldcrazyeye, 2024-08-01:
With the rise of toolchangers and awd custom machines I think this feature is extremely important to have working correctly and intuitively. Also need to ability to define multiple exclusion zones.

zruncho3d, 2024-08-01:
I'll throw my use case in: dual-gantry machines.

In the two-toolhead "Stay within the lines" mode, as well as one-toolhead "Just don't hit the other one" mode, exclusions with Manhattan moves to avoid head crashes (which can happen in the middle of the print) are pretty valuable.  Diagrams here:

https://github.com/zruncho3d/DuelingZero/blob/main/SOFTWARE.md

broncosis, 2025-01-31:
+1 would be great if we could have 3d model of the build chamber so things like tool docks could be excluded

closed 2025-05-09 by github-actions[bot]

lindnjoe, 2025-05-22:
This is an issue that shouldn't just be closed.  just saying.