OrcaSlicer/OrcaSlicer #6297: timelapse always enabled when print from sdcard

liciana24, 2024-07-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1. send print to printer and DESELECT the timelapse option
2. let print finish
3. using x1c touchscreen navigate to sd card and select print
4. notice missing timelapse option
5. print model
6. see timelapse recorded on sd card

### Actual results

when reprinting from printer timelapse option is MISSING and is ENABLED and saves timelapse to sd card

### Expected results

printer should show timelapse option when printing model sent from orca to printer. OR send gcode should DISABLE timelapse if it was not originally enabled

### Project file & Debug log uploads

all projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jschaefer-hub, 2024-09-18:
I encounter the same issue (Version 2.1.0)

Anthony-Bec, 2024-12-18:
I have the same issue too, it is the only issue that keeps me from moving from BambuStudio to Orca.

franciscomirasg, 2024-12-25:
I have the same issue too.

closed 2025-04-15 by github-actions[bot]