OrcaSlicer/OrcaSlicer #6300: Tree support always applies adaptive layer despite parameter setting

S1lentA0, 2024-07-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu P1S

### How to reproduce

When turning on treesupports, adaptive layer height is applied to the tree supports, despite turning this off in the slicer. Treesupports will break mid-print with printfailures as a result. 

After searching, [this seems ot be an old bug](https://github.com/SoftFever/OrcaSlicer/issues/5272), which still hasn't resolved.

### Actual results

As seen in the screenshots, gaps will form between the layer, because the slicer skips 1 layer for every 2 layers on the print itself.

[image]

[image]

### Expected results

By disabling adaptive layers for supports, I expect that this happens as intended.

### Project file & Debug log uploads

[2.zip](https://github.com/user-attachments/files/16437550/2.zip)
[log.zip](https://github.com/user-attachments/files/16437552/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

S1lentA0, 2024-08-01:
> Did you find any workarounds to make the supports work until this bug is fixed?

Adding more walls to the tree support seems to work. Last print the supports with 3 walls didn't fail.

tlhintoq, 2024-08-02:
There's two settings that do basically the same thing.  I couldn't tell ya what the difference is but they read the same and seem to do the same thing.  I always turn them both off and both on together.
[image]

S1lentA0, 2024-08-02:
> There's two settings that do basically the same thing. I couldn't tell ya what the difference is but they read the same and seem to do the same thing. I always turn them both off and both on together. 

What version are you using? I checked every version of tree support, but the option "independent support layer height" doesn't show up for me.

tlhintoq, 2024-08-03:
Currently 2.1.1 - but its been there a while

S1lentA0, 2024-08-03:
> Currently 2.1.1 - but its been there a while

Yeh, I just found out it will only appear if you have the Prime Tower disabled, which I never have. I made 4 slices comparing the settings and found out that the adaptive layer height does get disabled when disabling Independent Support Layer Height, and it still doesn't apply for Adaptive Layer Height, though there is difference in how the Tree Supports are generated. In this example I used Hybrid Tree Supports and all the settings, aside from these 4, remained the same.

[image]

[image]

[image]

[image]

Cilsor, 2024-08-03:
Which do you think would print the best? The very first almost looks as if you can just tell there is going to be delamination/warping on the trees. All the other trees look uniform and great. 

> > Currently 2.1.1 - but its been there a while
> 
> Yeh, I just found out it will only appear if you have the Prime Tower disabled, which I never have. I made 4 slices comparing the settings and found out that the adaptive layer height does get disabled when disabling Independent Support Layer Height, and it still doesn't apply for Adaptive Layer Height, though there is difference in how the Tree Supports are generated. In this example I used Hybrid Tree Supports and all the settings, aside from these 4, remained the same.
> 
> ![Both on](https://private-user-images.githubusercontent.com/134236161/354802835-40a8115a-9945-49dc-88d3-87b49e75ddd0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2OTU2MzIsIm5iZiI6MTcyMjY5NTMzMiwicGF0aCI6Ii8xMzQyMzYxNjEvMzU0ODAyODM1LTQwYTgxMTVhLTk5NDUtNDlkYy04OGQzLTg3YjQ5ZTc1ZGRkMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QxNDI4NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zM2U0ZTVhZWU5MjcyZmRjNWYwYzBjN2NkYjdkMGQ3MzE2MWE2N2YwNjZlOTU5NmIwYTUzMzE2NzI5MGNkM2NiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ZriwXjEdUDzmtSWXXksPQ-DXlF4rQSRSvkC2iQFNYDY)
> 
> ![independent off adaptive on](https://private-user-images.githubusercontent.com/134236161/354802838-808a9acf-f19c-4a9b-bba0-324754bb8eb9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2OTU2MzIsIm5iZiI6MTcyMjY5NTMzMiwicGF0aCI6Ii8xMzQyMzYxNjEvMzU0ODAyODM4LTgwOGE5YWNmLWYxOWMtNGE5Yi1iYmEwLTMyNDc1NGJiOGViOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QxNDI4NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMDRjNzgzMjFjNWI1MzZmNGM3ZTY3N2JjM2M0MTViYzEyMmQwN2Q2YjliNTIzODg2YmQ5M2IxNmMwYzUwOTc4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.hoOFWqCupQwgvPzfQ82-lSbE5HPBOyYkW1mREcYTIHI)
> 
> ![independent on adaptive off](https://private-user-images.githubusercontent.com/134236161/354802842-ab575cf4-7553-4a59-a4d6-61a3334a8db5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2OTU2MzIsIm5iZiI6MTcyMjY5NTMzMiwicGF0aCI6Ii8xMzQyMzYxNjEvMzU0ODAyODQyLWFiNTc1Y2Y0LTc1NTMtNGE1OS1hNGQ2LTYxYTMzMzRhOGRiNS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QxNDI4NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZGI4YmM0MGM0NTdlNTJlZGIzNzZlM2MxNjZjZmNhNzExMmU5MWNiYzIxYjU3NjQwNmNhNTQxZmFmZjZmMjliJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tDF0J_AZVUd4dLPXs53QH0IPWLcoXudBwkxN0yiEd9Q)
> 
> ![Both off](https://private-user-images.githubusercontent.com/134236161/354802846-5fd31a8f-2724-4541-9e5b-7e926cea9380.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2OTU2MzIsIm5iZiI6MTcyMjY5NTMzMiwicGF0aCI6Ii8xMzQyMzYxNjEvMzU0ODAyODQ2LTVmZDMxYThmLTI3MjQtNDU0MS05ZTViLTdlOTI2Y2VhOTM4MC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QxNDI4NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YTk3N2MxMTRmYmFhZTBhOTJmNGI1YWRlNzMyOThhOTg4NmJjYjA3NjBkZjM0OTAyNTEwMTYzNWVkOTc3NWU2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6rzRRddIhihvEpB7Zeu7b52c15oYnkJ0wl3TiM_MKcY)

closed 2024-11-10 by github-actions[bot]