OrcaSlicer/OrcaSlicer #6301: Easy: Do not go to Device Tab when print is sent to printer

Kingsidorak, 2024-07-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Clicking more than I would prefer

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Just a QoL thing for us simple farmers to not have to keep switching back to the Prepare or Preview tab

bhaal2, 2024-08-02:
Yes please!!

WimVDK, 2024-08-03:
Oh hell yes please, can't believe I didn't think of this. That default behavior is annoying as hell. Just because we send a print to the printer doesn't mean we wanna babysit it (especially if like in my case there printer is 2m away from the PC).

We're just go back to the prepare tab and start working on the next project

Jmaxime89, 2024-08-03:
Oh yes please ! This would be a great quality of life and would save me time at the end of the day

WimVDK, 2024-08-06:
you can already use the "send" option to send just the plate. image means nothing, unless i'm missreading that image

WimVDK, 2024-08-07:
Again, you an already use the 'send' option to send the sliced plate to the printer. That's not what this request is about. This request is after the print job has been sent to the printer (either via send or print option) that orcaslicer does not jump to the 'device' tab with the live stream. THAT'S what it's about, many of just want to start the job then go straight back to prepping the nice job, can't do that when you have to jump through hoops because the slicer defaults to the device tab

Kingsidorak, 2024-08-07:
It would also be very helpful if the sending could be done in the background so we can keep using the program

closed 2024-11-13 by github-actions[bot]