OrcaSlicer/OrcaSlicer #6305: More flow control

Stormwing01, 2024-07-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The problem and explination:

Currently you only have the total flow multiplier in the materials tab, and the top/bottom layers flow multiplier (that only tweaks the VERY top single layer and the very first, initial layer, nothing else).

However, in many of my prints using orca my walls themselves are heavily over-extruded, however internal solid infill is heavily UNDER extruded. this causes the top layers to create holes and structural weakness in my prints.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a flow control so I can adjust the flow for walls and internal solid infill separately. 
EG:
inner/outer wall flow of 0.99
Internal solid infill flow of 1.02

I believe that these extra flow settings could be a big benefit.

Troubleshooting to make sure it's no bug:
I have linear advance tuned. it makes zero difference if I set it to 0 or 0.7 (it just makes walls horrible at that high value.)
I am already printing slow at 40 mm/s and at 215 c with PLA.
My infill is gyroid, 15 percent.
the filament is dried in an active dryer at 50c for 8 hours beforehand to make sure there is no oozing.
ive also tweaked various other settings too, I can't remember all of them.

### Describe alternatives you've considered

There is literally no alternative. I have to live with both over and under extrusion simultaneously. if I increase flow, the print becomes a blob, and internal solid infill's 12 layers FINALLY close up a little. of I can turn down flow and suffer with massive holes in the top and bottom layers, (sue to internal solid infill causing top and bottom layers to droop and try to compensate for holes below them) but have smooth sides.

### Additional context

Look at the flow controls that cura has. it has inner wall flow, outer wall flow, infill flow, top flow, bottom flow... I can perfectly tune out any and all oddities and get beautiful prints, yet orca does not have these settings.

knw257, 2024-08-13:
While my issue is somewhat opposite from OPs, the request itself is desired. In my case, when I have an infill pattern that cross over itself on the same layer (e.g. triangle, tri-hex, rectilinear, etc), the nozzle scrapes lines it just laid down, leading to a nasty grinding noise. The ability to change the flow rate on sparse infill has been suggested as a way to reduce/eliminate this issue, which in turn helps reduce wear on consumables like the nozzle, and overall vibration on the machine.

Stormwing01, 2024-08-17:
> While my issue is somewhat opposite from OPs, the request itself is desired. In my case, when I have an infill pattern that cross over itself on the same layer (e.g. triangle, tri-hex, rectilinear, [...]

There really is a multitude of issues that can be fixed by giving the user controls. afterall, you and me are proof that two different people can have multiple different issues, even opposite issues, that can be fixed by giving us a few more settings.

Though to be fair, I think your issue can be fixed by just adding 0.2mm Z hop. if your nozzle is hitting the print that hard, it needs to be lifted during travel moves.

knw257, 2024-08-17:
> There really is a multitude of issues that can be fixed by giving the user controls. afterall, you and me are proof that two different people can have multiple different issues, even opposite [...]

It's not on travel moves. It lays down a line of infill, then on that same layer, a new line of infill is being laid down that intersects, and just runs right across it. As that progresses, those intersections build on top of one another. I'm actually surprised more people don't have this problem, because it's just inherent to the physical reality of infill patterns that self-intersect. 

I use tri-hex a lot for strength, but this is such a frustrating element of the pattern, and I'd love to reduce the flow rate to help account for this

Stormwing01, 2024-08-18:
> It's not on travel moves. It lays down a line of infill, then on that same layer, a new line of infill is being laid down that intersects, and just runs right across it. As that progresses, those [...]

Ohh, I see what you mean... Yeah, thats a very valid point, and fair obervation.

But yeah, Devs, if you guys are reading this... Please give us more/expanded flow controls...

TeenagedBurrito, 2024-11-13:
> It's not on travel moves. It lays down a line of infill, then on that same layer, a new line of infill is being laid down that intersects, and just runs right across it. As that progresses, those [...]

Orca slicer still has this issue and it seams bambu studio does a little a better with one less line. Ive been doing several flow calibrations with a custom cube and across every single test they all have the same line going across them on the 2nd to last layer. even on the lowest flow ive done you can vaguely see the line cutting across. Z hop is enabled. this happens when the nozzle travel to the start seam. Please a if developer see this could you add a zhop when travelling to z seam or zhop across solid and sparse infill

closed 2025-02-19 by github-actions[bot]

ColColonCleaner, 2025-06-09:
This would be lovely to have. I'm running into underextrusion of walls while infill is overextruded, and I can't adjust each flow independently.