OrcaSlicer/OrcaSlicer #6308: Weird slicing issue with 3DBenchy

alucryd, 2024-07-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Additional system information

CPU: AMD Ryzen 7800X3D
Memory: 32GB
GPU: AMD RX6900XT

### Printer

Creality Ender 3v3

### How to reproduce

Just slice the 3D Benchy STL.

### Actual results

Multiple hull lines are visible on the preview, and as expected end up highly visible on the print itself (not talking about the usual single hull line). I have never seen these lines in any other slicer, I've used Cura, SuperSlicer, PrusaSlicer before settling with Orca, and never had this issue. I tried different settings, an older issue was pointing at the `Ensure Wall Thickness` setting, but setting it to `None` doesn't help. I tried moving a few common settings, including `Classic` vs `Arachne` but never got rid of the lines. See screenshot.
[image]
Here are test prints, on the left Orca, on the right Prusa. Same filament, similar settings.
[image]

### Expected results

There should be no visible hull lines on the preview, and at most the usual one on the print itself.

### Project file & Debug log uploads

[3dbenchy.zip](https://github.com/user-attachments/files/16444853/3dbenchy.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Happy to do more tests as needed.

noadgskill, 2024-08-01:
The issue is with the seam location. Please manually change the seam to another location.

alucryd, 2024-08-01:
I was using scarf joint, that was the culprit indeed, thank you so much!

closed 2024-08-01 by alucryd