OrcaSlicer/OrcaSlicer #6315: AppImage Fails to start on Debian 12 Required libs missing

tjthorn, 2024-08-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.1.1

### Operating System (OS)

Linux

### OS Version

OS: Debian 12 bookworm

### Additional system information

OS: Debian 12 bookworm
Kernel: x86_64 Linux 6.1.0-22-amd64
DE: KDE 5.103.0 / Plasma 5.27.5
WM: KWin
CPU: AMD Ryzen 5 2600 Six-Core @ 12x 3.4GHz
GPU: AMD/ATI Ellesmere [Radeon RX 470/480/570/570X/580/580X/590]
RAM: 32024MiB

### Printer

n/a

### How to reproduce

Follwing work around from bug #6064 Install nightly build of OrcaSlicer_Linux_Ubuntu2404_V2.1.1.AppImage. Make Executable and execute from cli with ./OrcaSlicer_Linux_Ubuntu2404_V2.1.1.AppImage

### Actual results

Output to terminal upon execution of the reproduction steps:

/tmp/.mount_OrcaSldx30R1/bin/orca-slicer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_OrcaSldx30R1/bin/orca-slicer)
/tmp/.mount_OrcaSldx30R1/bin/orca-slicer: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by /tmp/.mount_OrcaSldx30R1/bin/orca-slicer)
/tmp/.mount_OrcaSldx30R1/bin/orca-slicer: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /tmp/.mount_OrcaSldx30R1/bin/orca-slicer)

### Expected results

App to launch and open on screen.sc

### Project file & Debug log uploads

Failure occurs prior to creation of $HOME/.config/OrcaSlicer 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tjthorn, 2024-08-02:
Yes 
```
ME@My-PC:~$ ldd --version
ldd (Debian GLIBC 2.36-9+deb12u7) 2.36
Copyright (C) 2022 Free Software Foundation, Inc.
```

bschatz, 2024-08-23:
OrcaSlicer_Linux_V2.1.1.AppImage works for me (but it crashes when i want to access the device)

As a remark:

``
... AppImage aims to be a format that is self-contained,  
and independent of the underlying Linux distribution ...
``

This is obviously not the case here, 
i would really would like that they would offer deb packages.

ozzyman, 2024-11-16:
Same issue with Ubuntu 22.04. 

Requires installation of Libwebkit2_4.1 and Glibc_2.38. Libwebkit2_4.1 is available, but Glibc_2.38 is not. The current version in the Ubuntu repository is Glibc_2.35. As bschatz notes, ALL dependencies should be included in the Appimage.

Jckf, 2024-11-21:
Same issue here, on Mint.

holla2040, 2024-11-28:
> Same issue with Ubuntu 22.04.
> 
> Requires installation of Libwebkit2_4.1 and Glibc_2.38. Libwebkit2_4.1 is available, but Glibc_2.38 is not. The current version in the Ubuntu repository is Glibc_2.35. As bschatz notes, ALL dependencies should be included in the Appimage.

same here,
 version `GLIBC_2.38' not found (required by /tmp/.mount_OrcaSlm5wjkC/bin/orca-slicer)

missingfield, 2025-01-04:
A bit late, but I also made this mistake. Note the filename's substring of "Ubuntu2404" communicates more recent version of glibc is required than is readily available on older distros.
Instead use the other appimage which, at time of writing, is OrcaSlicer_Linux_V2.2.0.AppImage

jbitterly, 2025-01-07:
I'm trying to run on Fedora kinoite, an immutable distro running KDE, and have the same issue with all released appimages.

"/tmp/.mount_OrcaSlYSQqBt/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory"

"ldd (GNU libc) 2.40"

This is an appimage, why isn't the dependency just packaged?

bepstein111, 2025-01-21:
> This is an appimage, why isn't the dependency just packaged?

Assuming whomever made it doesn't know/is learning how to do that.

Since it's such a widely used app, I would hope the fix is coming soon?

closed 2025-04-29 by github-actions[bot]

eliasmagn, 2025-10-04:
Still broken, its a joke! What a sad project!