OrcaSlicer/OrcaSlicer #6322: Z hop on a solid infill

LAZER555, 2024-08-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Sometimes when printer fills internal solid infill with any pattern, it fills for example one corner then moves another corner to fill but  it doesn't do z hop and scratches the surface. It may cause layer shift and/or scratch on a surface. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Would be cool to add otion to add z hops on a travel moves on solid infills.

Gork-Mork, 2024-08-06:
I have the same issue, my opinion this is a bug with the gcode. Orca Silcer doesn't care for what ever setting i try to toggle on or off. (Z-Hop is enabled as 0.4)
The issue is still there in OrcaSlicer_Windows_V2.2.0-dev_portable

File: [https://github.com/AdamV3D/Dayspring/blob/main/Mounts/Dayspring%20Mount%2045%20v12.stl](url) 
download the file and turn it as shown:
[image]

Example: The nozzle moves causing scratching on the infill surface, at layer 25
[image]

Photo from printed file with surface scratching.
[image]

The issue is not in Prusa Slicer and is behaving as expected with z-hop
[image]

LAZER555, 2024-09-21:
> I have the same issue, my opinion this is a bug with the gcode. Orca Silcer doesn't care for what ever setting i try to toggle on or off. (Z-Hop is enabled as 0.4) The issue is still there in OrcaSlicer_Windows_V2.2.0-dev_portable
> 
> File: [https://github.com/AdamV3D/Dayspring/blob/main/Mounts/Dayspring%20Mount%2045%20v12.stl](url) download the file and turn it as shown: ![bug1](https://private-user-images.githubusercontent.com/177685943/355605137-cbd162a2-c933-46e5-a311-b7ed4f7fd4b8.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTIwNDEsIm5iZiI6MTcyNjk1MTc0MSwicGF0aCI6Ii8xNzc2ODU5NDMvMzU1NjA1MTM3LWNiZDE2MmEyLWM5MzMtNDZlNS1hMzExLWI3ZWQ0ZjdmZDRiOC5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMDQ5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZjgyNTNhODdmOTgwOTBmNjllNDhhNDE1NDQ1NmJiZmZiYTcxMzI3MDFhMDk1ODA1OGUwOTljYjAzOGJhNWUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.moFB1oUL19CUISlQZiPBA0Ijyo9CJJxgOq17wjp1BgA)
> 
> Example: The nozzle moves causing scratching on the infill surface, at layer 25 ![bug2](https://private-user-images.githubusercontent.com/177685943/355605460-2802eff7-8167-4a23-9f8a-3fbda724c96b.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTIwNDEsIm5iZiI6MTcyNjk1MTc0MSwicGF0aCI6Ii8xNzc2ODU5NDMvMzU1NjA1NDYwLTI4MDJlZmY3LTgxNjctNGEyMy05ZjhhLTNmYmRhNzI0Yzk2Yi5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMDQ5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZjk1ODA5MWQ1ZjFhMzk1YjEyOTk0MTA5ZTBiNDg5NWY5MDE4OTdjN2RiYzg2YTcxN2Q5NWRmYWE0N2RiNzQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.MLjpL60LeG0O4eZ173Ha6FIbwlNqoC-4_F0lzEdAEIM)
> 
> Photo from printed file with surface scratching. ![PXL_20240806_212021737](https://private-user-images.githubusercontent.com/177685943/355606522-a864884f-11b7-4f0f-b818-d4db496c2531.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTIwNDEsIm5iZiI6MTcyNjk1MTc0MSwicGF0aCI6Ii8xNzc2ODU5NDMvMzU1NjA2NTIyLWE4NjQ4ODRmLTExYjctNGYwZi1iODE4LWQ0ZGI0OTZjMjUzMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMDQ5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNmE3OGNkMmZlYTExOWJiMDY4NWU4ODEwZDhmOGRiOGYyNzg5ZTY1ZmRiYTVhNzAzOTA3ZjZiOTUyNjI4ZjVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9vWzNdeiSrQqMAV-BYNlvJtG3cCWG3xaWppUDwZzUak)
> 
> The issue is not in Prusa Slicer and is behaving as expected with z-hop ![bug3](https://private-user-images.githubusercontent.com/177685943/355607526-a425f747-95b3-4c02-b75b-42db762d5a24.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NTIwNDEsIm5iZiI6MTcyNjk1MTc0MSwicGF0aCI6Ii8xNzc2ODU5NDMvMzU1NjA3NTI2LWE0MjVmNzQ3LTk1YjMtNGMwMi1iNzViLTQyZGI3NjJkNWEyNC5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMVQyMDQ5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZDQ5OTA0MjI1MGQwZjg4OTVhYjZjZWE4ZjViYTgxM2NkZjc4ODhiNDg0ODY1OTU0ZTgwYmVkNTYwNTVkNjhiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.lfauquDULFkXWnwDFFCKHk_Mted8lipBeJVr7MV6zjU)

I found this video https://www.youtube.com/watch?v=XTktr3zNXg8&list=LL&index=7&t=498s&ab_channel=Toffypops
i think it can help, but i haven't tested it yet

mdziekon, 2024-11-04:
@LAZER555  @Gork-Mork 

This behavior is controllable by config option **"Others > G-code output > Reduce infill retraction"**. I don't think it's a bug per-se, but I do believe it should be off by default, as it takes a while to figure out what's causing the "infill scratching".

closed 2025-02-10 by github-actions[bot]

arthur0071, 2025-10-06:
Need help on this, this is causing my parts to fail because the nozzle hits the infill(even gyroid 5%) and knocks out the little piece from the support, and i loose the print