OrcaSlicer/OrcaSlicer #6323: Unlink profile inherits

WimVDK, 2024-08-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This is likely tied to https://github.com/SoftFever/OrcaSlicer/issues/4691, since they didn't make a feature request, i am. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The ability to unlink preset or when saving as a new preset, no inherit ties are made

### Describe alternatives you've considered

There are no alternatives that i know off.

### Additional context

In orcaslicer with my Bambu X1C i always create a custom filament which i use as the basis (after printing a temp tower) to calibrate pressure advance and flow rate (in automatic mode, rarely fails on me). When saving the flow rate i'm asked to save it as a preset. Once tuned i save the file based on the nozzle installed (OBX for revo obxidian with my panda revo, RDB for revo diamond back,... since those values can differ).

This makes the original custom preset useless. The problem is when i try to remove the now obsolete preset i can't and i'm greeted with. If i'm missing something and is possible, please tell me how. But afaik it's not possible
[image]

Jmaxime89, 2024-08-03:
I'm glad to see I'm not the only one living with this issue. I hope this will be addressed

WimVDK, 2024-08-03:
Aye, it's gotten so annoying for me that i'm just gonna write my obx calibrated preset in excel and delete every single preset i have. Then make just OBX ones and make that my default. This behavior is so damn annoying

luksto, 2025-03-02:
+1 Still an issue!

pls256, 2025-06-04:
Really need this. Ability to either not inherit or resave without inheritance.

laserjet25, 2025-06-21:
This is a much needed for anyone with more than a single filiment/nozzle size. It is getting very annoying  having my preset list filled with a bunch of junk that I can't remove. Not sure the use case for this "feature".

Joshua-Beatty, 2025-07-01:
Would really like this, annoying to not be able to delete an unneeded preset because it is being inherited.