OrcaSlicer/OrcaSlicer #6326: Make "Top/Bottom surface flow ratio" and "Bridge/Internal Bridge flow ratio" settings clearer to understand

The760, 2024-08-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The "Top/Bottom surface flow ratio" and "Bridge/Internal bridge flow ratio" settings under the Quality tab are very confusing. It has been explained to me that Top/Bottom/Bridge are multiplied against the flow ratio of whatever filament you have selected. But then, Internal Bridge is multiplied against normal Bridge, which is then multiplied again against the filament flow ratio. 

As a system, I don't have issue with this. But OrcaSlicer does not explain this at all when you hover your mouse over these settings. How is the end user supposed to know how these functions work without clear messaging? When I first saw them, I thought it was completely separate numbers from the filament flow. I had no idea they were multipliers, because it wasn't stated as such.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Simply update the text when hovering over these settings to reflect that they are multipliers, and what they are multiplying against.

igiannakas, 2024-08-09:
Fixed in the above. Clarified all flow ratio tooltips (object flow ratio, filament flow ratio, bridge flow ratio, internal bridge flow ratio, Top surface flow ratio and Bottom surface flow ratio)

closed 2024-08-09 by SoftFever

Ergonomicmike, 2024-08-10:
@igiannakas If I may tack on here, when one performs the Flow Calibration test, is the "Top surface flow ratio" value ignored? (It is not temporarily set to 1.) As it appears now, two settings are affecting the Flow test - the "Master" Flow ratio, set in the Filament present, then the "Top surface flow ratio" set in Process.

Should I force "Top surface flow ratio" to 1 when performing the Flow Calibration test? (Or raise a separate Issue?)

igiannakas, 2024-08-11:
> @igiannakas If I may tack on here, when one performs the Flow Calibration test, is the "Top surface flow ratio" value ignored? (It does not temporarily set to 1.) As it appears now, two settings affect the Flow test - the "Master" Flow ratio, set in the Filament preset, then the "Top surface flow ratio" set in Process. If both, then it seems like I am chasing my tail, so to speak.
> 
> Should I force "Top surface flow ratio" to 1 when performing the Flow Calibration test? (Or raise a separate Issue?)

it depends what you want to achieve. If absolute flow calibration - set it to 1 (that is what I do). If you want to simulate what your model will look like leave it as is.

ideally I think the code should be setting it to 1 as you’re trying to calibrate the filament not the print profile but that’s just my view :)

Ergonomicmike, 2024-08-11:
Thanks. I agree that Top Flow should be set to 1 for the test. Otherwise, I could have it set to, say, 110% prior to the test, resulting in a nice looking top surface but anemic flow underneath.

I was wondering if Orca treats ALL the layers as one for this test - some kind of undefined "universal" layer - where all the flows are set to match the Flow Ratio in filament. But from what you're telling me, apparently it doesn't.

I think I'll make this a Feature Request.

igiannakas, 2024-08-11:
Nah don’t bother I can make the change fairly quick - will try and push a Pr on this today. It should be a couple of lines of code to set a default value

Ergonomicmike, 2024-08-11:
Thanks for your quick reply.

I thought about this more during my morning hike. I had totally forgotten that the proper way to set the "master" Flow ratio is by using the Califlower print. (Or [its equivalent on printables](https://www.printables.com/model/754877).) After that, top and bottom flow can be adjusted as needed.

So then, 1) The Calibration Flow test in Orca should really be renamed (or documented as) Top Layer Flow test Only. (To emphasize that it is only focused on the top layer.)

And 2) It would be okay to not reset the Top Layer flow to 1. If someone were simply rechecking their top flow and the test pattern shows that "0" is the correct value, then they don't have to change their top flow ratio from what they had previously set it to.

And if, say, the test shows to add 5%, then they need to multiply their current top flow ratio by 1.05, per the wiki.

Last, now that we have Small Area Flow Compensation, the wiki for Flow Calibration (the photos) might need to be updated a bit. Before Small Area Flow Compensation, the lumps, when over-extruding, where the infill reverses were fairly obvious. But with SAFC enabled, the over-extrusion at the reversals isn't as noticeable. See [Teaching Tech's photo](https://www.printables.com/model/904788-small-area-flow-compensation-tester#preview).

igiannakas, 2024-08-17:
Softfever is developing a new flow calibration that looks excellent. More here: 

https://github.com/SoftFever/OrcaSlicer/pull/6479

PR #6397 (2024-08-09): Clarified flow ratio tooltips
# Description

Clarified flow ratio tooltips to explain how the object flow ratio, filament flow ratio, bridge flow ratio, internal bridge flow ratio, Top surface flow ratio and Bottom surface flow ratio are related and used.

Fixes: #6326