OrcaSlicer/OrcaSlicer #6333: Support interface printed in air where no support is printed

knw257, 2024-08-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2

### Additional system information

CPU: 12th gen Intel i7-12700K
RAM: 32GB
GPU: AMD Radeon RX 6600

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

1. Import model base_v2.1.stl from https://www.thingiverse.com/thing:4920933/files
2. Lay on back of model
3. Enable supports
    - Tree(auto)
    - Tree Strong
    - On build plate only
4. Slice file

### Actual results

[image]

### Expected results

Support interfaces should only be printed where support exists beneath it. If support is prevented due to the 'On build plate only' restriction or similar, no interfaces should be printed there.

### Project file & Debug log uploads

[base_v2.1.3mf.zip](https://github.com/user-attachments/files/16483055/base_v2.1.3mf.zip)
[OrcaLogs.zip](https://github.com/user-attachments/files/16483057/OrcaLogs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-08-04:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/6259

knw257, 2024-08-05:
Is this a duplicate issue? While there are similarities between the two problems, it's not clear to me that they're caused by the same underlying issue.

tlhintoq, 2024-08-05:
To me it seems like it is.  Maybe two different manifestations of the same lack of checking.
They are both "support parts printing in mid air".  Whether that's just random bits of support in mid air, or interface in mid air... its still mid-air support printing.

All that's really happened though with that tag is a link is generated between the two tickets.  Whomever grabs one of the tickets will see the link.  While they work on 'A' they will at least be aware of 'B'.  Maybe it leads to a common cause, maybe it doesn't.  Maybe fixing one fixes the other for free, maybe it doesn't.  It doesn't close one ticket in favor of the other-Just makes a link on them for someone to be aware of.

knw257, 2024-08-06:
Thank you for the clarification, that makes total sense

tlhintoq, 2024-11-07:
Its not fixed - so its not stale.  Still a problem even in 2.2.0