OrcaSlicer/OrcaSlicer #6334: Add "Pause" option for pausing in the middle of a layer (bottom slider)

misterjmw, 2024-08-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am frequently annoyed when I have to add pauses mid-layer to a print, in-order to change filament colors because the slicer doesn't have the ability to do so.

### Which printers will be beneficial to this feature?

Klipper, Marlin, Others

### Describe the solution you'd like

I am still a lowly plebe with a single material printer. However, it's extremely useful to print in multiple colors. Example; I'm cutting text into a model and I want the background of the text to be contrasting and stand out once printed on top of. 

However, if I pause the layer, change filaments and then do the reverse for the next layer, the entire layer must be printed in that contrasting color. 

However **if** I could add a pause/filament change on the horizontal, mid-layer bar, I could trigger a pause before the section printed, then pause again after to swap back, thus only the feature I want is in the contrasting color.

While I think this pause is useful, I could alternatively see this as functionally being integrated into the 'paint' feature, so that you could paint a feature a color, and the slicer would group/organize pauses for it knowing that the printer profile does not have an MMU/AMS. 

### Describe alternatives you've considered

Alternatives considered; 
The only option for this (that I'm aware of) is to edit the gcode by hand, adding M25/filament change. However this is tedious because you have to know exactly where to pause.

### Additional context

This would help the majority of users who do not have MMU/AMS units.

iamapig120, 2024-08-04:
[image]
You could use this option to enable change fliament by yourself.

misterjmw, 2024-08-04:
@iamapig120 I do not follow how that achieves the goal. Does that allow the UI to add a 'filament change' anywhere? much like you can add a pause via right click, on the per-layer slider (on the right)?

chatasos, 2024-09-29:
Hi,

This is something i would also be interested in.

dakotawinslow, 2024-12-03:
I would also love to see this feature!

simpat1zq, 2025-01-15:
Just went searching to see how to do this and found this post. So I guess it answers my question that it's not possible. I'd love to see this as well.

bolens, 2025-02-27:
This feature would be really useful

Thomas49600, 2025-03-01:
Agreed, waiting around until my printer starts to print a purge tower where I then must manually pause and restart the print is a bit of a pain... especially if layers take a long time to print and I need to babysit the machine whilst I wait so that I don't miss the 10second pause window.

jnse78, 2025-03-18:
This would be a great function 👍

bigfoott, 2025-05-07:
I'd also love this!!

designnoevil, 2025-05-11:
yes adding a paurse mid layer, not just at new layer, would be very useful

aquapura001, 2025-05-29:
I'd love this feature too please. I need to be able to pause mid layer to remove threading when doing multi color.

Tolvelski, 2025-06-08:
I would also like this feature, but my reasoning is for calibration purposes. 

For example, I would like to be able to have a group of small, single layer squares arranged close together, and after each square, pull the hotend aside and present the bed. This would allow me to see my manual Z-Offset adjustment results more easily on my Neptune 4 Max, which has a large hotend that obstructs my view of my adjustments until the head has moved out of the way. This means I have to either manually pause the printer, or keep running the same print over and over. Now the print itself doesn't take too long, but with KAMP, Homing, Heatsoaking, etc, every new print adds up. 

Being able to have 9 squares, close together, print one, pause, present, adjust Z-Offset, and then resume, would be nice.

Ineedmoney420, 2025-09-03:
I know tool changers I becoming more popular but I see myself wanting this on multiple occasion having only a single tool at the moment..

christopher-lambe, 2026-04-03:
Workaround for mid layer pause for magnet insertions

Yes we all know how - insert a pause (right click on + in vertical slider) at the start of the layer that covers the magnet (the layer where the bridge is printed - the hole is shown as closed) and click slice again. But this is not good enough. The nozzle oozes and loses pressure the outside walls may suffer quality issues.

THE WORKAROUND

Set Timelapse to Smooth: This moves the head out of the way (so it not all over the place in the timelapse) and also importantly makes a prime tower. The nozzle is primed at the start of each layer. So you can insert the pauses as needed and any oozing is taken care of when the nozzle is primed on the tower. Just be quick when inserting the magnets so the print does not cool too much and you cause a hull line.

christopher-lambe, 2026-04-03:
> Yeah, just searched for this feature for the same purpose: wanted to pause mid-layer to drop some magnets in it.

Workaround above - sorry forgot to quote you for my reply

Incendos, 2026-04-03:
> > Yeah, just searched for this feature for the same purpose: wanted to pause mid-layer to drop some magnets in it.
> 
> Workaround above - sorry forgot to quote you for my reply

It is an acceptable workaround, but when i have a 250mm tall model printing and i want to drop magnets near the top, i don't really want to waste ~30g for a prime tower. I rather let the infill take the loss.

misterjmw, 2026-08-25:
Keeping this alive. Still waiting for a mid-layer pause feature.