OrcaSlicer/OrcaSlicer #6336: Place seams only on vertices option

Ataraxia-Mechanica, 2024-08-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

New feature in Cura 5.8, which allows constraining seams on vertices. This would make seam painting easier, also makes random/nearest seam on inorganic objects look better.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Constrain seams on vertices

Ataraxia-Mechanica, 2024-11-03:
pls no stale

DARKFiB3R, 2024-11-14:
This would be nice