OrcaSlicer/OrcaSlicer #6338: where is the wall thickness information like in prusa slicer

moth4017, 2024-08-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

in prusa slicer when you set number of perimeters it calacltes the wall thickness where is this in orca slicer
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

[image]

igiannakas, 2024-08-04:
Second tab, first setting.

moth4017, 2024-08-04:
Hi it does not give you the wall thickness measurement  as it does in prusa slicer as per my screen shot.

E.g  2 walls = 1.59 mm  wall thickness

We use this wall thickness measurement  when designing  in CAD

Regards

Mark

igiannakas, 2024-08-04:
Can you post a screenshot of the said setting in prusa slicer?

moth4017, 2024-08-05:
[cid:1082ddce-8214-46b0-aa95-d3ca804ab598]

moth4017, 2024-08-10:
i dont want to open up prusa slicer  to get the dimensions, please could you add this info back in , thank you

closed 2024-11-16 by github-actions[bot]