OrcaSlicer/OrcaSlicer #6341: 100% AUX fan not respected

psiberfunk, 2024-08-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

MacOS Sonoma 14.6

### Additional system information

N/A

### Printer

Bambu X1C

### How to reproduce

Slice Project file attached, see that M106 P2 S255 does not get set where it should in the GCODE (also attached).  AUX fan stays at 0% for the entire print, which is not desired.

Set the filament aux fan cooling rate to 99% and note that M106 P2 S252 **_does_** get properly set.   There seems to be an issue with the 100% setting for some reason.

### Actual results

Aux fan does not come on when it should after the first layer.

### Expected results

Aux fan SHOULD come on at 100%  after the first layer.

### Project file & Debug log uploads

Project File and GCODE output zipped here:
[BadAuxFanExample.zip](https://github.com/user-attachments/files/16487967/BadAuxFanExample.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Reproducible.

mark-mcdougall, 2024-08-08:
Thanks for the tip re setting it to 99%, I’m having the same issue on my p1 and this is a good work around so in the interim

SoftFever, 2024-08-10:
It's bug of "Fan speed-up time" setting.
This is feature is a bit problematic with it's current implementation.
 I suggest turn it off for now

closed 2024-11-16 by github-actions[bot]

psiberfunk, 2024-11-16:
Is this really not planned ?