OrcaSlicer/OrcaSlicer #6348: Very slow slicing speed on Ubuntu compared to PrusaSlicer

Leonti, 2024-08-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04

### Additional system information

CPU: Ryzen 7 2700X 3.7 GHz 8-Core Processor
GPU: Nvidia GTX 1050 (driver version: 535.183.01)

### Printer

Voron Trident

### How to reproduce

1. Add an STL to a plate, I used [front_skirt_a_300.stl](https://github.com/VoronDesign/Voron-Trident/blob/main/STLs/Skirt/300/front_skirt_a_300.stl)
2. Slice the plate using `0.20mm Standard @Voron` system preset by pressing `Slice plate` button and stopwatch on the phone
3. Press stopwatch end once slicing is finished
4. Slice the same model on PrusaSlicer with a similar profile and capture slicing speed with the same stopwatch

### Actual results

The place is finished with OrcaSlicer in *8s*
The plate is finished in *1-2* seconds with PrusaSlicer

### Expected results

Similar slicing speeds. The effect is even more pronounced with big plates, sometimes I have to wait for a minute or even more to get the plate sliced.

### Project file & Debug log uploads

[debug_Mon_Aug_05_18_58_31_110998.log.zip](https://github.com/user-attachments/files/16493355/debug_Mon_Aug_05_18_58_31_110998.log.zip)
[front_skirt_a_300.zip](https://github.com/user-attachments/files/16493356/front_skirt_a_300.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I realize that this kind of problems are hard to debug and probably system-specific, so I guess at this stage I'm looking for ways to provide more relevant information.
I run it with `./OrcaSlicer_Linux_V2.1.1.AppImage --debug 5`, but logs don't seem to include anything related to performance.

tlhintoq, 2024-08-06:
Are your supports set to `Tree | Strong` ?
I have an open ticket for Tree strong taking 24hrs on a big job while the same slice using Tree|Default takes 10mins

Leonti, 2024-08-06:
> Are your supports set to `Tree | Strong` ? I have an open ticket for Tree strong taking 24hrs on a big job while the same slice using Tree|Default takes 10mins

I don't have supports enabled, but I tried different supports for testing just now and the time was still around 8s for the same model.

As another data point I tried downloading OrcaSlicer 2.1.1 on another machine, also running Ubuntu 22.04 and had very similar performance results.

closed 2024-11-13 by github-actions[bot]