OrcaSlicer/OrcaSlicer #6351: Walls loops around holes

Alkerion, 2024-08-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

On some models I want to strengthen the holes with more walls around them.
Actually with modifiers it's quite complicated and cumbersome to perform as there is no simple option for this.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

To have an option to set the wall count (loops) around holes.

### Describe alternatives you've considered

use modifiers, but it's lot of work and result is not always good.

zingo, 2024-08-05:
This is a great idea to get extra walls around holes when using inserts and not needing to bump all walls :)

digital0785, 2024-08-09:
good idea!

Alkerion, 2024-08-22:
More than annoying with the awful unusable coordinates system still not been fixed.

And not clean as the slicer add the extra walls around the modifier, on top and bottom, not just the hole, so you have to be precise.
[image]
[image]
[image]

Would talk 0.5s to do this with a simple setting, and will be clean.

TylonHH, 2024-08-24:
https://github.com/SoftFever/OrcaSlicer/issues/5832

xokahag, 2024-10-19:
Instead of increasing wall count, use 100% concentric infill for the modifier. This prevents this double outer wall issue on your last screenshot.

closed 2025-01-25 by github-actions[bot]