OrcaSlicer/OrcaSlicer #6353: ABS is missing for 0.4 nozzle for Adventurer 5m

Openwrtfunboy, 2024-08-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

Latest Dev and latest stable 2.1.1

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04

### Printer

Flashforge adventurer 5m

### How to reproduce

1. Select adventurer 5m with 0.4 nozzle
2. Abs is missing
3. But for 5m pro (same except closed case and camera) there is abs profile

### Actual results

No ABS profile for 0.4 nozzle

### Expected results

ABS profile :D

### Project file & Debug log uploads

There is no need on it

### Checklist of files to include

- [X] Log file
- [X] Project file

Avpman2, 2024-08-05:
I don't think you should be printing ABS in a non-enclosed 5M. It's toxic. Maybe that's why they left it out?

Openwrtfunboy, 2024-08-06:
> I don't think you should be printing ABS in a non-enclosed 5M. It's toxic. Maybe that's why they left it out?

I don't fucking care what you think.

There is abs profile for 0.25 nozzle, but its missing for 0.4

Some peoples smoke some people print ABS, let me decide what to do with my life.

Avpman2, 2024-08-06:
I used the phrase not aimed at you; I should have said: "we" or something similar.
Classy response, by the way.

closed 2024-11-13 by github-actions[bot]

InactiveCoyote, 2026-04-16:
also need help