OrcaSlicer/OrcaSlicer #6361: BBL bind with pin code doesn't work

lin-ycv, 2024-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11 22H2

### Printer

Bambu Lab P1S (FW:01.06.01.00)

### How to reproduce

1. Sign out of any account on the printer
2. Click bind with pin code in orca
3. Click submit

### Actual results

nothing happens (tried with LAN mode on and off)

### Expected results

printer should ask if I want to bind

### Project file & Debug log uploads

[debug_Tue_Aug_06_12_49_02_11592.log.zip](https://github.com/user-attachments/files/16505957/debug_Tue_Aug_06_12_49_02_11592.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Elchschmied, 2024-08-25:
Hello, my X1C gives an 8 digit hex code and in Orca I have only 6 digits. Don't know what is the right code.
Thank you

Hlaxfors, 2024-09-20:
I have the same issue.

Hlaxfors, 2024-09-20:
are you logged into a BBL account on Orca?

lin-ycv, 2024-09-23:
I tried both signed in and signed out, neither worked. 
I think it's a networking issue.
My desktop is connect to my main router via lan.
My printer is connected to an isolated AP. 
My desktop is connected to the AP via wireless. 
Thru orca, I'm unable to find and bund the printers. But I can run a python script to force it to connect. 
On my laptop that's connected only to the AP via wireless, orca can find the printer just fine.

momentmal, 2024-09-30:
same problem for me on MacOS. I get the PIN prompt, i go to the printer, logged out, read out the PIN from account and enter it, press OK, nothing happens.

momentmal, 2024-09-30:
ok, just realized I have to login in the "Home" Section with my Bambulab Account..

thanks reddit

https://www.reddit.com/r/BambuLab/comments/1ec77dz/how_to_connect_printer_to_orca_slicer/

closed 2024-12-16 by lin-ycv