OrcaSlicer/OrcaSlicer #6365: The Orca slicer does not work on Linux Mint 22.

dkrast, 2024-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.1.1

### Operating System (OS)

Linux

### OS Version

Mint 22

### Additional system information

HP Zbook I9

### Printer

Qidi Q1 Pro

### How to reproduce

The Orca slicer does not start on Linux Mint 22.

### Actual results

Nothing

### Expected results

In Mint 21 Work O.K.

### Project file & Debug log uploads

[debug_Mon_Jul_01_21_34_42_3642.log.0.zip](https://github.com/user-attachments/files/16515774/debug_Mon_Jul_01_21_34_42_3642.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

imacmill, 2024-08-08:
Confirmed.

When running Orca Slicer AppImage from a command prompt, the following error results:

"/tmp/.mount_OrcaSlKi1X8M/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory"

The repositories for Mint 22 contains no such version of that lib, so manual install of the lib fails.

imacmill, 2024-08-09:
> Bambu Studio already released a additional version especially for Ubuntu 24.04 (what mint 22 based on) Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage
> 
> This version is working fine
> 
> I think we have to wait until a new compiled version is ready.

Thank you...that version works.

pimmja, 2024-08-11:
> Bambu Studio already released a additional version especially for Ubuntu 24.04 (what mint 22 based on) Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage
> 
> This version is working fine
> 
> I think we have to wait until a new compiled version is ready.

Is there a link to this AppImage available?

imacmill, 2024-08-11:
> > Bambu Studio already released a additional version especially for Ubuntu 24.04 (what mint 22 based on) Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage
> > This version is working fine
> > I think we have to wait until a new compiled version is ready.
> 
> Is there a link to this AppImage available?

https://github.com/bambulab/BambuStudio/releases

Note that Bambu Studio does not appear to support Klipper. I couldn't find a way to directly send sliced projects to my printer over the network.

BadPixel89, 2024-08-16:
I didn't realise there was a dedicated thread for mint.
This is how I worked around the issue: 

In basic terms, create a link to the new libs with the name of the old links, create a start menu shortcut to include an argument that seems to be required to see parts of the UI.

https://github.com/SoftFever/OrcaSlicer/issues/425#issuecomment-2292341048

A new appimage would be a better option but that works for me in the meantime

dprossner, 2024-08-19:
try adding the printer manually and then logging in to bambu account. should show custom configs after that

tcables, 2024-09-27:
> Relates to: [4132](https://github.com/SoftFever/OrcaSlicer/issues/4132)
> 
> setup Dialog is blank and errors in console present for me in mint 21.3 and orca 2.1.1.
> 
> > Found that you can run:
> > `WEBKIT_DISABLE_COMPOSITING_MODE=1 /PathToAppImage`
> > in a shell.
> > Works for me with latest stable build 2.1.1
> > Thanks to Post by [ahrifox950](https://forums.linuxmint.com/memberlist.php?mode=viewprofile&u=383068)
> > https://forums.linuxmint.com/viewtopic.php?p=2516253&sid=62f3db3956c845404ce782301220421d#p2516253
> 
> [#4132 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/4132#issuecomment-2378207515)

closed 2025-01-03 by github-actions[bot]