OrcaSlicer/OrcaSlicer #6366: Separate speed settings for crossing and non-crossing infill

Ataraxia-Mechanica, 2024-08-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

New feature in prusa slicer 2.8.0, the speed for crossing and non-crossing infill can be set differently. Maybe also make this true for acceleration?
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add this feature so it's a bit more convenient when switching infill types

igiannakas, 2024-08-09:
Isn’t the same thing achieved by just slowing down the sparse infill print speed? What am I missing?

Ataraxia-Mechanica, 2024-08-10:
This is about e.g. setting non-crossing infill's speed to 400, crossing's speed to 300, then just by changing the infill type from rectlinear to cubic, the sparse infill's speed get reduced automatically to 300, saving a bit of time.

closed 2024-11-16 by github-actions[bot]