OrcaSlicer/OrcaSlicer #6367: Actual speed preview

Ataraxia-Mechanica, 2024-08-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Very cool new feature in Prusaslicer 2.8.0
[image]
Add this pls

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add speed preview so it's more clear how the printer moves exactly

kylek29, 2024-08-15:
Just as a side note, Creality's new slicer is based on OrcaSlicer and I believe has the PrusaSlicer's new speed visualization ported to it, so any dev looking to port that feature in to Orca may find that a tad easier since it should better align code wise.

ckgdmm1, 2024-09-16:
Doesn't seem to be in the 2.2 Beta release ;(

SphaeroX, 2024-09-20:
This could indeed be very helpful to optimize and fine-tune your 3D printing. I hope this feature will be included so that problems with new features can be better identified.

Ataraxia-Mechanica, 2024-12-20:
pls no stale

ckgdmm1, 2025-04-07:
I asked Igiannakas he said it will probably be in the next release, this was a month ago

Ataraxia-Mechanica, 2025-05-28:
> Any news?

Can't find anything related in pull requrest page / Igiannakas' profile, I think we are cooked

SphaeroX, 2025-12-07:
@as-com How is your pull request coming along? Do you think you will get it merged? This feature is really missing in OrcaSlicer.