OrcaSlicer/OrcaSlicer #6374: Dark Mode doesn't work on Linux appimage.

NoSadBeHappy, 2024-08-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Fedora 40

### Printer

Bambulab A1

### How to reproduce

Launch `OrcaSlicer_Linux_V2.1.1.AppImage` on fedora 40 with desktop theme set to dark mode
see that orca slicer is not in dark mode as it should be.

### Actual results

orca slicer is in light mode

### Expected results

orca slicer should be in dark mode

### Project file & Debug log uploads

[debug_Wed_Aug_07_15_15_37_89457.log](https://github.com/user-attachments/files/16534747/debug_Wed_Aug_07_15_15_37_89457.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Another solution would be to add a checkbox for dark mode in the preferences menu like is on windows.

Dawoox, 2024-11-12:
Up, still a relevant issue with 2.2.0

nikoraasu, 2024-11-22:
Unstable bin package from AUR (2.2.0.rc-1) also has this issue, despite "Breeze-dark" being enabled, light theme is present

flickerfly, 2025-02-01:
I'm using 2.2.0 and trying to find out if there is a dark mode and how to enable it. Seems like the checkbox for it is missing in the preferences section.

[image]

flickerfly, 2025-02-01:
On Fedora 40 using AppImage, I can get Dark Mode if I start orca with this command `GTK_THEME=Adwaita:dark orca`.

Prazgreen1, 2025-08-19:
> still an issue with orcaslicer 2.3.0, but I can't reopen the issue.

I have an enhancement idea in the github issues for this repo for a dark mode toggle on the Linux version which would be a good solution to avoid any inconsistencies in different tooling and theming. Adwaita for example is very different from Breeze.