OrcaSlicer/OrcaSlicer #6376: Preview: Nozzle Positioning Error (Nozzle Floating in Dead Space instead of on top of Model)

xiexie-ah, 2024-08-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.0

### Printer

Flashforge Adventurer 5M Pro

### How to reproduce

Slice any model, go to preview, and slide the bottom nozzle preview slider and you'll notice that the nozzle is off to the side instead of on the model

### Actual results

https://github.com/user-attachments/assets/0fa93fa5-cb82-4422-912a-80e0954c25bb

### Expected results

Expect nozzle to be situated above the actual model

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

noadgskill, 2024-08-10:
This issue is caused by Flashforge violating the Orca license and modifying it arbitrarily. Flashforge printers must use Flashforge-Orca. This issue is something that Flashforge should solve, not Orca.

closed 2024-11-10 by xiexie-ah

yw4z, 2025-09-21:
fixed in https://github.com/SoftFever/OrcaSlicer/pull/10806