OrcaSlicer/OrcaSlicer #6378: Orca is leaving a wide gap in the seam

HikiWaltz, 2024-08-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Device name	HikiSRV
Processor	AMD Ryzen 9 7900 12-Core Processor                3.70 GHz
Installed RAM	64.0 GB (63.1 GB usable)
Graphics: RTX A4500 20gb

### Printer

Elegoo Neptune 4 Max / Neptune 4 Pro / Creality K1Max

### How to reproduce

Just add 2 STL's with thin walls (I've tested with 3, 4, 6 walls)
Seam gap of 0.1, 0.05, 0.01, 0.
Retraction works fine for any other print and even for the same STL if it's alone on the plate.

### Actual results

[image]
[image]

### Expected results

Normal closed seams

### Project file & Debug log uploads

[Plates2.zip](https://github.com/user-attachments/files/16537804/Plates2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-08-08:
Do you have the same issue in other slicers?

HikiWaltz, 2024-08-08:
> Do you have the same issue in other slicers?

I've tested on creality print and cura... 
Both printed the seams correctly.
What I find most weird is that I've just printed the same stl's, no change to the configs at all, but one at a time.
Both printed perfectly.

It really only happens when printing more than one STL in the same plate.
They even print normally, aside from one or both having this gap at the seam.
Printing more than 2 always give me gaps in each one.

igiannakas, 2024-08-08:
The below can cause what you're seeing:
1. Your extra length on restart is set to 0.5 which means you are extruding more than retracting (0.2). If that is needed **you have an issue with your retraction settings** - most probably it's too low so **you are leaking material out of the nozzle while traveling** and then you need to unretract more to compensate which isn't always successful hence the gap.
2. Your seam gap should be expressed as a %age. **Start by setting it to 0%** (without the % you are setting absolute gap values)
3. Your **PA value is not tuned correctly** - try calibrating this again.
4. Try **traveling faster and with a faster acceleration**. This allows for less time for the nozzle to leak.

In general this issue happens either due to incorrect setting of the seam gap (too large) or because your nozzle is leaking material during travel or due to wrong PA. To combat the leaky nozzle (especially with a 0.6 nozzle) you need higher retraction values and faster travel speeds. The other two are just calibrations /settings that just need to be set correctly.

HikiWaltz, 2024-08-09:
Just Tested.
It improved a lot, but still happening here and there.
Removed restart length and increased retraction, no significant change on it's own.
Increasing the travel speed seems to be the most effective.
Using gap seam at 0% leaves the seam too fat. Using 5~7% to make it similar to 0.05.
Can't see why you say my PA is not tuned correctly, tho... I'm getting pretty sharp corners.

Still... out of 6 machines (Usually printing with same gcode configs), 2 beds had 1 stl each with open gaps... Still trying to investigate further causes.
Bit wonky due to deadline and having to "test in prod" kkkk

igiannakas, 2024-08-09:
If it was me I’d combine the fastest travel speed you can do together with a bit higher retraction. From what you’ve said it appears to be due to nozzle ooze. Maybe also try slightly lower filament print temperature (like 5-10c) to reduce how “runny” it is.

ericaprieto, 2024-10-22:
This is definitely related to multiple objects and rounded corners. I printed the same box, with the same settings, multiple times, always get this when I have more copies at the same, funny thing: I tried 3 copied, 1 was perfect and the other 2 have these same games. Watching it closely it looks like it only starts extruding after it does the curve on that corner.

staal54a, 2024-10-27:
I'm seeing some very similar issues on my X1C with there being a seam gap despite adjusting the likely culprit parameters: set seam gap to 0 and 0%, dried filament thoroughly, switched between arachne and classic, tuned PA manually and used the auto calibration. None of that seems to work.

The only difference for me is that I didn't need to attempt to print multiple parts to see the issue. The STL I was printing, however, did have a lot of thin walls and it seemed like a few layers were not as problematic as others. I also tested printing a single wall cylinder and the seam was okay, but there were certainly slightly larger gaps at various heights.

The other interesting part is I have so far not been able to recreate it on any of my other printers (all bedslingers); I only see it on the X1C. I've only tried to recreate the single wall cylinder at this point though (will attempt to test the problematic STL when I get a chance).

grrminator, 2024-11-22:
Can confirm. I set seam gap to 0, 2.5, 5, 7.5 and default 10 (all percents) and the gap is still the same. All exactly the same witdth. A mm or two wide.

I already have PA calibrated etc.

grrminator, 2024-11-22:
This was not an issue previously in orca 1.8 or so. Now it is and IDK why after updating.

staal54a, 2024-12-02:
Here is a simple test case I use which recreates the seam issue I'm seeing with the X1C. I'm wondering if others can run this test and also recreate it (should be relatively quick and use minimal filament):

1. In OrcaSlicer, open a new project
2. Add a cylinder via Right Click -> Add Primitive -> Cylinder
3. Either in the global print settings or the per object settings, update the following:
    - Set wall loops to 1
    - Set infill density to 0%
    - Set top layers to 0
    - Set bottom layers to 1
4. Then, in the filament overrides settings, ensure that Retract on Layer Change is enabled
5. (Optional) Maybe disable arc fitting, I am able to recreate the issue with it but it does result in different gcode so might be worth just removing that variable

Print the file and not if you are still seeing the same seam gap

staal54a, 2024-12-05:
I've played around with various settings and I still haven't identified the issue or found what I would consider a good solution (I've messed with a lot of different factors one at a time and to various extremes). At this point in time, I think my best path moving forward is to use the Extra Length on Restart for deretraction moves. I do get slightly better results lowering my retraction to an absurdly low amount (0.05mm as one example) but that does cause a little bit more stringing on retraction test models. Plus I'm using a Bambu X1C whose base 0.8mm retraction distance is somewhat dialed in.

Just as a note, I think the reason a few people above said they only see it with multiple items could be related to the retract on layer change setting. For my one wall cylinder test print, if I disable that setting the cylinder prints fine as long as there is just one since that removes the retraction that would normally occur at the seam. However, with multiple objects on the plate, that retraction would still occur before a travel move.

closed 2025-03-13 by github-actions[bot]

igiannakas, 2025-03-31:
This is a printer tuning issue vs a slicer code issue. If the gcode shows the seam correctly starting and ending, the printer is the one that needs tuning to accurately perform the moves instructed by the slicer I’m afraid.

KizzyCode, 2025-04-01:
Ok, for me the problem resolved itself: For some reason, slice gap and adaptive pressure advance were not honored (or I did something very wrong) – after recreating the profiles with the same settings it works now.

fbeltrao-br, 2025-12-31:
I have an Orbiter V2 extruder with a Mosquito Magnum hotend in direct drive configuration.

I noticed this issue with Orca migrating from Superslicer and Prusa only when printing multiple objects. Same printer, I tried to keep the same settings across all slicers. This hollow cylinder primitive with a single wall is enough to reproduce the issue:

[image]

What I tried:
- Increase or disable retractions: increasing it from default 0.2mm (usually ranges form 0.2mm to 0.5mm depending on the filament type) to 1mm makes no visible change. Disabling it causes stringing and blobs.
- Increase or reduce printing speeds: increasing it just causes visible resonance on external layers, and reducing it makes no visible change.
- Reduce or disable seam gap: no visible change.
- Disable pressure advance: default is 0.045, no visible change, just rounded corners.
- Uninstall and delete all slicer data in appdata, to recreate the profile.

The only thing I found to make a difference:
- Configure a retract_restart_extra of 0.1mm: this closes the seam in the object that would show the issue, **but on the other object the seam becomes a little fatter.**

I feel like the detract is not happening when nozzle travels between the first objects to the second, or it's being applied twice, causing the gap.

doctorbede, 2026-05-30:
Wooooooooooooow! Wa wee wa!

This may actually be the breakthrough.

Go into your Filament Settings and make sure all retraction overrides are turned OFF. If those overrides are enabled, Orca will ignore the retraction settings from your printer profile, which means all the changes you've been testing may never have been applied in the first place.

Once the filament overrides are disabled, retraction control is returned to the printer settings and your adjustments will finally take effect.

So if you've been changing retraction settings and seeing little or no difference, this is likely why.

Now that you have proper control again, you can finally start dialing in the seam gap problem with confidence.