OrcaSlicer/OrcaSlicer #6382: Opening file from thingiverse issue

GTreaper, 2024-08-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win 10

### Printer

Voron 2.4 350

### How to reproduce

1. go to https://www.thingiverse.com/thing:6076779/files
2. press open in orca slicer
3. receive orca slicer warning "The download has failed:
You are not allowed to use robots, spiders, scraping, or other technology to access or use
Content or any Sites to obtain any information"

### Actual results

[image]

### Expected results

open file in orca 

### Project file & Debug log uploads

cant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TheDuke22, 2024-08-09:
Same problem happen to me on Arm Mac (same orca version)

boromyr, 2024-08-10:
same problem

NuclearCreations, 2024-08-10:
Same problem.

ZAKsPop, 2024-08-11:
Yeah, what gives? Orca had the option to do this in previous version which I thought was cool but now it doesn't work.

astanix, 2024-08-13:
I can confirm that this is happening for me as well.

awdev1, 2024-08-14:
Can confirm this is happening here.

Faellatv, 2024-08-14:
Did the same for another file, same problem here.

3DCoded, 2024-08-14:
Same problem here. It looks like it might be Thingiverse that is blocking the download?

awdev1, 2024-08-14:
Maybe the user agent orca slicer uses?

eacxx, 2024-09-04:
Another one bites the dust...
Orca 2.2.0_dev
Mac ARM, Sonoma 14.6.1

closed 2024-12-11 by github-actions[bot]

pickanothername, 2024-12-11:
This has been resolved for me. M3 Macbook Pro Sonoma

awdev1, 2024-12-17:
Working now @GTreaper can close