OrcaSlicer/OrcaSlicer #6390: Custom printer and filament settings are not imported for my Bambu Lab P1S

electricar, 2024-08-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

 OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev.AppImage 

### Operating System (OS)

Linux

### OS Version

Linux Mint 22 Cinnamon

### Printer

Bambu Lab P1S

### How to reproduce

1. Start the application.
2. Finish the "Get Started" form.
3. Install the Bambu Lab network plugin.
4. Select to import custom printer and filament settings.
5. Search the custom printer and filament settings in the prepare tab.

### Actual results

My custom printer and filament settings are not visible/imported.

### Expected results

My custom printer and filament settings are visible/imported.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16553605/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

discip, 2024-08-09:
@electricar
Duplicate of #6365.
Please close in favor of the mentioned issue report. 😊

electricar, 2024-08-09:
@discip 
I don't think that it is a duplicate. 
It is true that the stable 2.1.1 release does not start on Linux Mint 22, but the nighly version "OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev.AppImage" does start.

Only the import of my custom printer and filament settings from my Bambu Lab profile does not work.

EightFinger, 2024-08-15:
Same problem here, newest nightly build as an appimage. Can launch orcaslicer but it doesn´t import my existing config using the Bambu account. I am on Linux Mint 22.

dprossner, 2024-08-19:
try adding the printer manually and then logging in to bambu account. should show custom configs after that

electricar, 2024-08-20:
> try adding the printer manually and then logging in to bambu account. should show custom configs after that

isn't this the normal workflow already?

EightFinger, 2024-08-21:
As I have Voron Trident, maybe its different from a P1S, but i can´t synchronize my existing presets even if I add a printer beforehand,

closed 2024-11-27 by github-actions[bot]