OrcaSlicer/OrcaSlicer #6391: Solid Infill Combination

ZedOud, 2024-08-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I was inspired by this [video](https://www.youtube.com/watch?v=ss610c_KxKc) about taking grams of plastic and printing time out of infill for strengthening parts (except for that which is strictly necessary to support top surfaces) and using it to thicken the shell instead.

I have developed a routine for simple parts to print thicker bottom and top layers quickly. My default print heigh is 0.15mm with infill combination enabled, however internal solid infill is also printed at 0.15 instead of 0.3. (0.14-0.16 is the optimal height for strength on a 0.4mm nozzle.)

So I use an object modifier that overlaps the bottom layers and top layers with settings set as follows:
- 1-2 top/bottom layer
- 100% infill
- monotonic infill pattern

This basically achieves internal solid infill printed at infill combination layer height.

It would be nice to have a checkmark to do this for all top/bottom surfaces, especially if we can avoid 1-2 overhanging surfaces so the combination layers are well supported.

One problem I anticipate is using this with layer heights <33% of the nozzle width (like 0.1mm with a 0.4mm nozzle). This will result in the infamous problem of greater than 80% nozzle width layer heights. Imagine: a layer height of 0.4mm on a 0.4mm nozzle for internal solid layers. The whole point of printing more internal solid infill for strength, and this would ruin the layer adhesion.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A checkbox to enable this feature and (optionally) an integer for how many top/bottom layers should be combined away, or else how many should be retained for overhang/bridging and top surface handling.

Also, for anyone who prints with layer heights below 1/3 of the nozzle width it would be necessary to provide some sort of cap on the number of layer combinations.

### Describe alternatives you've considered

The current alternative available is to manually drop lots of modifier objects to target all the top/bottom layers you want to convert to this style of combined internal solid infill.

### Additional context

Also, #5421 here is the latest enhancement issue asking for a way to control the number/multiplier on combination layers.

discip, 2024-08-09:
@vovodroid
Maybe you could take on this task since you have already contributed a whole bunch of very useful features! 👍

thanks in advance 😊

discip, 2024-08-09:
@vovodroid
Maybe you could take on this task since you have already contributed a whole bunch of very useful features! 👍

This might help as well: https://github.com/SoftFever/OrcaSlicer/issues/5421#issuecomment-2253636173

thanks in advance 😊

igiannakas, 2024-08-09:
Raised a PR that aims to address this above.

vovodroid, 2024-08-09:
It's not the same - FR author would like to apply infill combination also for solid.

igiannakas, 2024-08-09:
Ah misunderstood. I’ll untag it.

Ataraxia-Mechanica, 2024-08-11:
I remember this being a thing in some random obscure slicer from a printer company. Surprised to find out the slicer supports setting how many solid infill layers get combined, and how many sparse infill get combined separately, but doesn't support changing line width.

igiannakas, 2024-08-11:
The issue I see with combining solid layers is how do you deal with non rectangular models.

Basically, if you print a curved model with say 0.1 LH and you combine solid infill every 3 layers, the overlap between the solid infill and the walls will change as the angle of the surface/walls of the model changes. I think this may end up leading to benchy hull line type artefacts unless the model is perfectly flat. 

For sparse infill this is not a problem as the area is mostly void and the overlap is far less visible due to the density of the infill being low. But for solid infill you’ll get a “rectangle” over multiple layers trying to fit a slice of a “sphere” if that makes any sense. 

Like imagine the below: 
[image]

lines on the side are a cross section of the 0.1LH for perimeters with the block in the middle being the cross section of the solid infill line combined over 4 layers.

Ataraxia-Mechanica, 2024-08-11:
I've printed at 0.05 height with 0.4mm nozzles before and it would be useful in that situation. Probably too niche and isn't worth implementing due to the disadvantage you listed?

closed 2024-11-17 by github-actions[bot]