OrcaSlicer/OrcaSlicer #6392: Add Alpha channel to Color Selector

Fonzi03, 2024-08-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The colors available to assign to filaments can only be solid due to the lack of an alpha channel. When printing transparent filaments it would be nice to be able to set the alpha channel to better represent them, especially when additional colors can be on top of the transparent filament. 

### Which printers will be beneficial to this feature?

Klipper, Marlin, Others, All

### Describe the solution you'd like

Add another slider next to the Lum slider, but for Alpha, add an integer input field for alpha as well. The integer box and slider should mirror each other.

Fonzi03, 2025-02-07:
Bump to keep alive

dstulken, 2025-04-07:
I'll second this feature request.

Linking some other requests/discussions:
https://github.com/SoftFever/OrcaSlicer/issues/2431,
https://github.com/SoftFever/OrcaSlicer/discussions/3884
https://forum.bambulab.com/t/how-to-define-filament-color-as-transparent/15299/18

yw4z, 2026-10-17:
implemented in https://github.com/OrcaSlicer/OrcaSlicer/pull/16197

closed 2026-10-17 by yw4z