OrcaSlicer/OrcaSlicer #6393: Filament for Features

RNGIllSkillz, 2024-08-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-dev

### Operating System (OS)

Windows

### OS Version

win 10

### Printer

voron 2.4

### How to reproduce

Try to print with different filament for infill

### Actual results

gcode with only primary filament

### Expected results

gcode with multiple filaments

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/16555879/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Also it seems that filaments in Filament for Features fields does not change when filament is changed in the filament section.
[image]

ProtomakerSprint, 2024-08-09:
I wanted to 3D Print using PETG filament so I select the SOVOL PETG option
but when I slice it says PLA in the filament which is my previous 3D Filament setting; saving as Gcode shows the Printer temperatures are still set to PLA.
Why can I not see temperatures ?

I am used to Cura with Marlin

obertini78, 2024-09-18:
If you go to Process->Objects, there you can change the Filament for Features and it works.

https://github.com/user-attachments/assets/d9756710-1700-4d83-a8e7-4fa3d91bfbf4

OrcaSlicer 2.2-beta

closed 2024-12-26 by github-actions[bot]