OrcaSlicer/OrcaSlicer #6399: Orcaslicer will not boot on Pop!_OS 24.04 (Cosmic DE)

Firnthon, 2024-08-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Pop!_OS 24.04 alpha

### Printer

QIDI X-Max 3

### How to reproduce

1. Download AppImage
2. Make executable
3. Open in terminal
4. Fix standard libwebkit problem with `sudo ln -s /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37;sudo ln -s /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18`
5. Slicer is not booting with error `[2024-08-09 09:19:46.459508] [0x00007a221bab13c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:5567): Gdk-WARNING **: 09:19:51.343: The program 'orca-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadWindow (invalid Window parameter)'.
  (Details: serial 3763 error_code 3 request_code 4 (core protocol) minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)`

### Actual results

Orcaslicer is not loading

### Expected results

Load the slicer.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Firnthon, 2024-08-09:
To fix this problem X window app should use Wayland by default. This is not working. To force Orca using Wayland in the terminal type `unset $WAYLAND_DISPLAY` and then tun Orca from terminal.

paelnever, 2024-08-17:
> To fix this problem X window app should use Wayland by default. This is not working. To force Orca using Wayland in the terminal type `unset $WAYLAND_DISPLAY` and then tun Orca from terminal. If this will not solve the problemu - reboot OS.

So the app only works under wayland? Not possible support for systems without wayland?

GruffyPuffy, 2024-08-18:
I tested with Cosmic DE (wayland) on Pop!_OS 24.04 and got the same as above, I also have GNOME added to this install and when starting a GNOME (wayland) session I do not get this problem. Just added here in case it helps developers to understand. Also used the dev-release of OrcaSlicer which is made for Ubuntu 24.04 (new version for WebKit) and it is the same, however the symlink-trick is not needed.

agravelot, 2024-08-25:
Linked to https://github.com/pop-os/cosmic-epoch/issues/466

GruffyPuffy, 2024-09-03:
Updated my Cosmic DE today, after that I can now start OrcaSlicer (the U24.04 version).

closed 2024-12-11 by github-actions[bot]