OrcaSlicer/OrcaSlicer #6406: Command Line args not working on Windows

tyleranton, 2024-08-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Run OrcaSlicer via command line with arguments: `./orca-slicer.exe --help`

### Actual results

No output
[image]

### Expected results

It should output help flags

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I've also tried piping the output to a file, but the file remains empty: `./orca-slicer.exe --help > out.txt`

A similar [issue](https://github.com/SoftFever/OrcaSlicer/issues/3132) was opened in December of last year, but it went stale without response. Is this just not meant to be supported on Windows or is this a bug?

hYdos, 2024-09-18:
this is still an issue

rizwansarwar, 2024-12-30:
Have same problem as above. --help works on Linux version of same release but not on windows.

closed 2025-04-27 by SoftFever