OrcaSlicer/OrcaSlicer #6416: Displaying true printer speeds

ckgdmm1, 2024-08-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hard to tune filament for high speed printing when it's unclear whether nozzle is reaching desired speeds or being capped by acceleration

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

Would be super cool if one of the displays showed true printing speeds that take into account acceleration

### Additional context

Big fan of Orcaslicer, have bought you guys 2 coffees on kofi already

DVSVIDEO, 2024-08-10:
I think the printer firmware is responsible for measuring and displaying the speed.
The slicer software does not display anything. It slices the model and that's it. The task from there is the printer's job.

michmela44, 2024-08-10:
If you mean in the slicer, does #6367 "Actual Speed Preview" being ported from PrusaSlicer accomplish what you are after?

ckgdmm1, 2024-08-10:
@DVSVIDEO Not quite sure what you mean

ckgdmm1, 2024-08-10:
@michmela44 YES, this is exactly what I'm looking for. Do you recon this is something they would be interested in porting over? It helps that PrusaSlicer is open source...

michmela44, 2024-08-10:
No idea, I'm just a potential user. Just pointing it out since that one was already filed, that maybe it would be better to consolidate and comment/upvote that request to gain more visibility.

I would assume they would almost certainly accept a PR from someone implementing in, but have no idea about their current workload/time constraints, etc.

closed 2024-11-16 by github-actions[bot]