OrcaSlicer/OrcaSlicer #6423: Easy Skirt Removal

Dumplap, 2024-08-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In the attempt of reducing waste on the skirt prime I only print with one layer although this is often requires tools to remove when the PEI is midly hot.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

add a small blob on the 3rd or 4th layer (skipping the second) providing an easy point to remove the skirt.

vovodroid, 2024-08-18:
> in the attempt of reducing waste on the skirt prime I only print with one layer

Do you think one or two more skirt loops are really big waste?

Dumplap, 2024-08-19:
> > in the attempt of reducing waste on the skirt prime I only print with one layer

> Do you think one or two more skirt loops are really big waste?

why waste more than we have too?

closed 2024-11-26 by github-actions[bot]