OrcaSlicer/OrcaSlicer #6431: Bambu X1C webcam crashes Orca Slicer since new release!

rayman2386, 2024-08-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1 Carbon

### How to reproduce

Since updating to 2.1.0 and 2.1.1, the webcam hasn't worked in Orca Slicer and crashes the application every single time. It works perfectly fine in Bambu Studio and I've tried both the network plugin provided by reinstalling Bambu Studio and the network plugin that Orca Slicer provides.

### Actual results

Despite trying both network plugins provided by Bambu Studio and Orca Slicer, trying to use the webcam in Orca Slicer crashes the application every single time.

### Expected results

In previous updates like 1.9, reinstalling the network plugin provided by Bambu Studio fixed this issue allowing the webcam to work in Orca Slicer, though in that version the application didn't crash, it just gave an error message and failed to load the webcam. It doesn't work with the network plugin provided by Orca Slicer either.

### Project file & Debug log uploads

[First Prints.zip](https://github.com/user-attachments/files/16576437/First.Prints.zip)
[log.zip](https://github.com/user-attachments/files/16576444/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-08-14:
Suggestion:

- Update Orca to current 2.2.0 nightly
- Update BambuSlicer so you have the latest shared parts for communication.

[image]

xokahag, 2024-10-19:
Does your system have an integrated and dedicated GPU by any chance? On Windows there's a bug with the video feed driver that crashes it if you use the dedicated GPU. The only workaround for this is to force windows to always use the internal GPU when running Orca. You can find the setting here: System > Display > Graphics > select Orca > Options

closed 2025-01-25 by github-actions[bot]