OrcaSlicer/OrcaSlicer #6433: LINUX - WAYLAND - NO 3D VIEW AFTER NVIDIA DRIVER UPDATE

deddroid, 2024-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Cachyos

### Printer

FLSUN SR with Speederpad (klipper)

### How to reproduce

open the .appimage

### Actual results

can't see nothing in the right border of the screen (where the stl have the 3d view)

### Expected results

normally the screen load every panels

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

THE LOG:

[2024-08-12 11:42:16.015638] [0x00007c8c48b9ab80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.524: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.524: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:16.524: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.524: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.524: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.525: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:16.525: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.525: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:16.525: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:16.525: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[... 296 lines cut ...]

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:18.815: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:18.815: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Failed to create GBM buffer of size 1567x1099: Invalid argument

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:18.822: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:18.822: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:18.823: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:18.823: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:18.827: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:18.827: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:18.829: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:18.829: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:16702): GLib-GObject-CRITICAL **: 11:42:18.829: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:16702): Gtk-CRITICAL **: 11:42:18.829: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

nicman23, 2024-08-14:
i have been dealing with this as well. slicing works and even uploading the gcode, even if you cannot see anything

spusuf, 2024-08-15:
Same issue with nvidia 560 on arch with 4070ti

fivetide, 2024-08-15:
same, but with nouveau, not proprietary drivers. Could it be a wayland issue? 

```
python-pywayland 0.4.18-1
qt5-wayland 5.15.14+kde+r59-1
qt6-wayland 6.7.2-4
wayland 1.23.0-1
wayland-protocols 1.36-1
xorg-xwayland 24.1.2-1
xwaylandvideobridge 0.4.0-2

linux 6.10.4.arch2-1
linux-api-headers 6.10-1
linux-firmware 20240809.59460076-1
linux-headers 6.10.4.arch2-1
linux-lts-headers 6.6.45-1
syslinux 6.04.pre2.r11.gbf6db5b4-4
util-linux 2.40.2-1
util-linux-libs 2.40.2-1

vulkan-headers 1:1.3.285-1
vulkan-icd-loader 1.3.285-1
vulkan-intel 1:24.1.5-2
vulkan-mesa-layers 1:24.1.5-2
vulkan-nouveau 1:24.1.5-2
vulkan-tools 1.3.269-1
vulkan-validation-layers 1.3.275-1
```

GyrasS, 2024-08-15:
tried it on x11 with the same result, so it's more likely to do with the gpu drivers

nicman23, 2024-08-16:
i switched to a different vt and startx. it worked fine there, so it is not an general issue with orca but rather something with wayland.

i am running self compiled master

deddroid, 2024-08-17:
is there a workaround?

deddroid, 2024-08-17:
usually for this type of problems i prefer flatpak packages, but orca didnt have a flatpak version

Nathan22211, 2024-08-17:
what DE are you running? because there's a lot of different wayland implementations out there right now

deddroid, 2024-08-18:
I am on KDE Plasma 6.1.4

ereenke, 2024-08-19:
I have the same situation. Switching from Wayland to X11 fixes the problem.

CachyOS linux (arch based)
KDE 6.1.4
NVIDIA 560.31.02

ktw, 2024-08-19:
Same here:
Fedora 40, 64 bit
Gnome 46 / Wayland
NVIDIA 560.31.02
Card: 1080 TI

deddroid, 2024-08-19:
from "journalctl -xe" this is what I read:

"Unità UNIT entrata nello stato 'failed' (fallito) con risultato 'exit-code'. ("unit UNIT is in failed state(failed) with 'exit-code' result")
ago 19 20:44:05 cachyos-x8664 dolphin[135691]: kf.config.core: Access to "/home/cachy/Apps/squashfs-root/OrcaSlicer.desktop" denied, not owned by root and>
ago 19 20:44:05 cachyos-x8664 dolphin[135691]: kf.config.core: Access to "/home/cachy/Apps/squashfs-root/OrcaSlicer.desktop" denied, not owned by root and>
ago 19 20:44:30 cachyos-x8664 kernel: orcaslicer_main[144957]: segfault at 6fdf ip 00005ec81e43ae74 sp 00007ffd024dc758 error 4 in orca-slicer[3be9e74,5ec>
ago 19 20:44:30 cachyos-x8664 kernel: Code: 0f 83 48 ff ff ff 44 89 d3 eb bf 0f 1f 00 48 03 45 10 e9 d3 fe ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40>
ago 19 20:44:53 cachyos-x8664 wireplumber[1113]: wp-event-dispatcher: <WpAsyncEventHook:0x63627298cc50> failed: <WpSiStandardLink:0x636272beeb20> link fai>
ago 19 20:44:55 cachyos-x8664 kwin_wayland[134599]: kf.windowsystem: static bool KX11Extras::mapViewport() may only be used on X11
ago 19 20:44:56 cachyos-x8664 kwin_wayland_wrapper[134599]: src/gbm_drv_common.c:131: GBM-DRV error (get_bytes_per_component): Unknown or not supported fo>
ago 19 20:44:56 cachyos-x8664 kwin_wayland_wrapper[134599]: src/gbm_drv_common.c:131: GBM-DRV error (get_bytes_per_component): Unknown or not supported fo>
ago 19 20:44:56 cachyos-x8664 kwin_wayland[134599]: kwin_scene_opengl: 0x502: GL_INVALID_OPERATION error generated.
ago 19 20:44:56 cachyos-x8664 kwin_wayland[134599]: kwin_scene_opengl: 0x502: GL_INVALID_OPERATION error generated.
ago 19 20:44:56 cachyos-x8664 kwin_wayland[134599]: kwin_scene_opengl: 0x502: GL_INVALID_OPERATION error generated.
ago 19 20:44:56 cachyos-x8664 kwin_wayland[134599]: kwin_scene_opengl: 0x502: GL_INVALID_OPERATION error generated.
ago 19 20:44:56 cachyos-x8664 kwin_wayland[134599]: kwin_scene_opengl: 0x502: GL_INVALID_OPERATION error generated.
ago 19 20:44:56 cachyos-x8664 kwin_wayland[134599]: kwin_scene_opengl: 0x502: GL_INVALID_OPERATION error generated.
"

deddroid, 2024-08-19:
from "journalctl -xe" this is what I read:

"Unità tmp-.mount_OrcaSllmdoUY.mount entrata con successo nello stato 'dead' (morto).
ago 19 20:54:17 cachyos-x8664 kwin_wayland[134599]: kf.windowsystem: static bool KX11Extras::mapViewport() may only be used on X11
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kf.windowsystem: static bool KX11Extras::mapViewport() may only be used on X11
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kf.windowsystem: static bool KX11Extras::mapViewport() may only be used on X11
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: Failed to focus 0x4c00037 (error 3)
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: Failed to focus 0x4c00037 (error 3)
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: Failed to focus 0x4c00037 (error 3)
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: XCB error: 3 (BadWindow), sequence: 12380, resource id: 79691831, major code: 129 (SHAPE), >
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: XCB error: 3 (BadWindow), sequence: 12381, resource id: 79691831, major code: 129 (SHAPE), >
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: XCB error: 3 (BadWindow), sequence: 12394, resource id: 79691831, major code: 129 (SHAPE), >
ago 19 20:54:18 cachyos-x8664 kwin_wayland[134599]: kwin_core: XCB error: 3 (BadWindow), sequence: 12395, resource id: 79691831, major code: 129 (SHAPE), >
ago 19 20:56:24 cachyos-x8664 systemd[997]: Reload requested from client PID 146083 ('systemctl') (unit app-org.kde.konsole@17dfee98acb04d88b399a6a212970f>
ago 19 20:56:24 cachyos-x8664 systemd[997]: Reloading...
ago 19 20:56:24 cachyos-x8664 systemd-xdg-autostart-generator[146089]: /home/cachy/.config/autostart/discover_overlay.desktop: stat() failed, ignoring: No>
ago 19 20:56:24 cachyos-x8664 systemd-xdg-autostart-generator[146089]: /home/cachy/.config/autostart/discover_overlay_configure.desktop: stat() failed, ig>
ago 19 20:56:24 cachyos-x8664 systemd[997]: Reloading finished in 166 ms.
ago 19 20:56:24 cachyos-x8664 kernel: orcaslicer_main[146095]: segfault at 6fdf ip 00005a16cc3dbe74 sp 00007ffe3c35a7b8 error 4 in orca-slicer[3be9e74,5a1>
ago 19 20:56:24 cachyos-x8664 kernel: Code: 0f 83 48 ff ff ff 44 89 d3 eb bf 0f 1f 00 48 03 45 10 e9 d3 fe ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40>
ago 19 20:56:24 cachyos-x8664 systemd[1]: tmp-.mount_OrcaSluNEbkP.mount: Deactivated successfully.
░░ Subject: Unità terminata
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ Unità tmp-.mount_OrcaSluNEbkP.mount entrata con successo nello stato 'dead' (morto).

"

amdavidson, 2024-08-31:
Same for me, happened with the 560 driver update and is consistent on Arch and Fedora.

This issue also exists on the bambu studio side of things: https://github.com/bambulab/BambuStudio/issues/4669

deddroid, 2024-08-31:
with last nvidia driver update it seems tro work again 

nvidia driver: 560.35.03
cachyos
kde 6

nicman23, 2024-09-01:
no it does not.... arch linux 560.35.03

xPakrikx, 2024-09-01:
Same problem on Fedora 40.

amdavidson, 2024-09-01:
arch / nvidia-dkms 560.35.03 / plasma 6.1-1 / wayland still has the same issue.

downgrading to nvidia-dkms 555.58.02-1 corrects it again.

spusuf, 2024-09-08:
Since this is an issue that is present in upstream BambuSlicer, I checked their issue repo and the solution is to add the following environment variables to enable software rendering (until the bug is resolved):
``` __GLX_VENDOR_LIBRARY_NAME=mesa ```
and 
```__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json```

Tested working on Arch with Nvidia 560.35.03 on GNOME wayland.

mikezila, 2024-09-10:
That does work for me on Arch with the latest Nvidia drivers. KDE 6, Wayland.

nicman23, 2024-09-10:
wtf how does that even make sense? does it switch llvmpipe?

harry42203, 2024-10-19:
> Since this is an issue that is present in upstream BambuSlicer, I checked their issue repo and the solution is to add the following environment variables to enable software rendering (until the bug [...]

How do I set these, will it be set for just orca or is it generally enabling software rendering for everything?

amdavidson, 2024-10-19:
if you edit your .desktop file you can add the environment vars to just orcaslicer. Attached is an image of the editor in Plasma.
[image]

ronb1964, 2024-11-03:
This worked for a while on Nobara but now I'm back to having to log into an X11 session. I'm on Gnome 46.6 using NVidia 565.57.01. Not the end of the world I suppose but a pain to have to log out/in to use the software. I'm new with Github so if I can help in any way I'd be happy to. Thank you all for your efforts.

mabl, 2024-12-18:
On nixos, I managed to get it to use software rendering using

```
__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/run/opengl-driver/share/glvnd/egl_vendor.d/50_mesa.json orca-slicer
```

nicman23, 2025-01-04:
a better solution is to start Xephyr (you still get 3d accel that way)

after you setup virtualgl

you do

```
Xephyr -ac -br -noreset -screen 1920x1080 :100 & sleep 0.1;
DISPLAY=:100 WEBKIT_DISABLE_DMABUF_RENDERER=1  orca-slicer
```

nicman23, 2025-01-04:
a way that maintains 3d accel (no software rendering) is:

install virtualgl

run vglserver_config and select the first option, allow all users etc

run a Xephyr ie `Xephyr -ac -br -noreset -screen 1920x1080 :100`

run the slicer `export DISPLAY=:100; vglrun -d egl orca-slicer`

jameskitt616, 2025-01-04:
For the https://github.com/bambulab/BambuStudio/issues/4669 Issue, it looks like there is a solution to get it running with the flatpak version again. It seems to be working when adding the environment variable to bambulab via e.g. flatseal `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json` eventho this file doesnt exist. https://github.com/bambulab/BambuStudio/issues/4669

nicman23, 2025-01-04:
it just falls back to software rendering

dasmikko, 2025-02-04:
> a way that maintains 3d accel (no software rendering) is:
> 
> install virtualgl
> 
> run vglserver_config and select the first option, allow all users etc
> 
> run a Xephyr ie `Xephyr -ac -br -noreset -screen 1920x1080 :100`
> 
> run the slicer `export DISPLAY=:100; vglrun -d egl orca-slicer`

Makes performance a lot better! Is there a way to make the orca slicer window bigger? I can't seem to resize the application. 
I can resize the Xephyr window.

nicman23, 2025-02-05:
use this `Xephyr -ac -br -noreset -screen 1920x1080 :100 &>/dev/null & sleep 1; DISPLAY=:100 vglrun openbox & DISPLAY=:100 WEBKIT_DISABLE_DMABUF_RENDERER=1 vglrun orca-slicer`

QQuark, 2025-02-16:
For anyone interested, on my side with gnome on wayland, with Xwayland running (which I think is started by gnome ?), just running `DISPLAY=:0 vglrun orca-slicer` seems to work fine without needing Xephyr (adding the WEBKIT variable thing makes the whole thing segfault within a minute or so).
Acceleration is definitely working, as it was ultra slow with the MESA workaround, and now it runs great.

zomgugoff, 2025-02-16:
On Arch/Plasma 6.3/Wayland, I had to leave the DISPLAY variable off to get it to work, but vglrun does the job.

jusii, 2025-02-18:
I've had so much headache with Nvidia, multiple monitors (4 currently) and slicers in Linux,I thought to post some success, finally. Was using X11, because 4 monitors just didn't work with Wayland. Now I found Nvidia 570 drivers out (or as beta in popos) and downloaded those. I have RTX 3500 Ada.

Well, atleast for 10 minutes 4 monitors working great and even OrcaSlicer is accelerated! (couldn't get that working in X11) Still have to start with this command line: __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json OrcaSlicer

mrehkopf, 2025-02-25:
For me, this combination of incantations works well, with hardware acceleration, no missing 3D view, UI elements, or recent files view, on Fedora 41/Plasma 6.3/Wayland:

`WEBKIT_DISABLE_COMPOSITING_MODE=1 vglrun ./OrcaSlicer_Linux_AppImage_V2.3.0-beta.AppImage`

EntwicklungsAnstaltGretsch, 2025-03-14:
I had the same issue and disabling graphics acceleration stopped working.

This solution partially worked for me. I am on Arch Linux (Garuda KDE to be precise):

> a way that maintains 3d accel (no software rendering) is:
> 
> install virtualgl
> 
> run vglserver_config and select the first option, allow all users etc
> 
> run a Xephyr ie `Xephyr -ac -br -noreset -screen 1920x1080 :100`
> 
> run the slicer `export DISPLAY=:100; vglrun -d egl orca-slicer`

i did install virtualgl but am starting orca-slicer like this:

`vlgrun orca-slicer %F`

you may have to replace `orca-slicer` with whereever your program is setup

This works for me and also has hardware acceleration so the gcode preview is not laggy any more like it was with doing the `env WEBKIT_DISABLE_COMPOSITING_MODE=1` for startup, but rather buttery smooth like you would expect it to be.

Hope this helps

zomgugoff, 2025-03-22:
I'm now seeing a behavior in 2.3.0 where the program will not start with vglrun until after it is run once without it. I assume it's setting something up that vglrun prevents.

zomgugoff, 2025-04-08:
Commit 32efc17 is working properly without vglrun

nicman23, 2025-04-09:
the fix for me for both webkit and 3d view was to buy a 9070 xt lmao

johne8, 2025-05-14:
(Fedora 41, hyprland (wayland), Nvidia 575.51.03)
I run it in a script like this:
```
#!/bin/bash

export __GLX_VENDOR_LIBRARY_NAME=mesa
export __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
export MESA_LOADER_DRIVER_OVERRIDE=zink
export GALLIUM_DRIVER=zink
export WEBKIT_DISABLE_DMABUF_RENDERER=1

./OrcaSlicer_Linux_AppImage_V2.3.0.AppImage
exit $?

```

Only issue i have is that the videostream from my printer is black.

nullcubee, 2025-05-14:
Oh that works so much better than purely software rendering! Thank you, @johne8!

For my NixOS brethren who use Home Manager, here's a configuration that will add a desktop file that can be used by whatever desktop launcher you have:
```nix
xdg.desktopEntries = {
  OrcaSlicer = {
    name = "OrcaSlicer";
    genericName = "3D Printing Software";
    icon = "OrcaSlicer";
    exec = lib.concatStringsSep " " [
      "env __GLX_VENDOR_LIBRARY_NAME=mesa"
      "__EGL_VENDOR_LIBRARY_FILENAMES=${pkgs.mesa}/share/glvnd/egl_vendor.d/50_mesa.json"
      "MESA_LOADER_DRIVER_OVERRIDE=zink"
      "GALLIUM_DRIVER=zink"
      "WEBKIT_DISABLE_DMABUF_RENDERER=1"
      "${getExe pkgs.orca-slicer} %U"
    ];
    terminal = false;
    type = "Application";
    mimeType = [
      "model/stl"
      "model/3mf"
      "application/vnd.ms-3mfdocument"
      "application/prs.wavefront-obj"
      "application/x-amf"
    ];
    categories = [
      "Graphics"
      "3DGraphics"
      "Engineering"
    ];
    startupNotify = false;
  };
};
```

alexhorner, 2025-06-22:
https://github.com/bambulab/BambuStudio/issues/4669#issuecomment-2923389879

> Cross-post from [#6951](https://github.com/bambulab/BambuStudio/issues/6951):
> 
> For the Flatpak, the Mesa files seem to have moved around. You can get normal hardware acceleration on NVIDIA by passing
> 
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/50_mesa.json
> GALLIUM_DRIVER=zink 
> MESA_LOADER_DRIVER_OVERRIDE=zink 
> WEBKIT_DISABLE_DMABUF_RENDERER=1
> 
> via `--env` to `flatpak run` (i.e., not just as ENV generally)
> 
> All in for pasting:
> 
> ```
> --env=__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/50_mesa.json --env=GALLIUM_DRIVER=zink --env=MESA_LOADER_DRIVER_OVERRIDE=zink --env=WEBKIT_DISABLE_DMABUF_RENDERER=1
> ```
> 
> Basically the vendor files seem to be under `/usr/lib/x86_64-linux-gnu/GL/glvnd/` now.

This works for me on Fedora 42 with Flatpak via Flatseal

zomgugoff, 2025-07-04:
As of build a8141ef, program is crashing on startup again. Terminal shows a bunch of gtk resize errors, the splash screen is displayed, then the main windows opens very briefly and the projects screen is visible before it crashes. Back to vglrun...

ryodeushii, 2025-07-04:
> As of commit [a8141ef](https://github.com/SoftFever/OrcaSlicer/commit/a8141ef360092aae9f2d93399ef6a3d359e8a11e), program is crashing on startup again. Terminal shows a bunch of gtk resize errors, the splash screen is displayed, then the main windows opens very briefly and the projects screen is visible before it crashes. Back to vglrun...

I was able to make it run with software renderer and XWayland
```
env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 GBM_BACKEND=dri /usr/bin/orca-slicer
```

you can try this (if the path to orca-slicer binary is the same or you can just leave orca-slicer without a path - I have original path on my machine here because I did set up shell alias for it).
But basically I replaced orca-slicer part in `/usr/share/applications/OrcaSlicer.desktop` Exec entry with this snippet.

ryodeushii, 2025-07-04:
upd: `env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 GBM_BACKEND=dri /usr/bin/orca-slicer` this is actually allowing whole slicer to work and even Klipper ui (reported in several issues) is not crashing device tab

smooth-baking-powder, 2025-12-06:
> ```
> [Desktop Entry]
> Name=OrcaSlicer
> GenericName=3D Printing Software
> Icon=io.github.softfever.OrcaSlicer
> Exec=/bin/bash -c '__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_DISABLE_DMABUF_RENDERER=1 XDG_SESSION_TYPE=x11 GBM_BACKEND=dri /usr/bin/flatpak run --branch=master --arch=x86_64 --command=entrypoint --file-forwarding io.github.softfever.OrcaSlicer @@u %U @@'
> Terminal=false
> Type=Application
> MimeType=model/stl;model/3mf;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;x-scheme-handler/orcaslicer;
> Categories=Graphics;3DGraphics;Engineering;
> Keywords=3D;Printing;Slicer;slice;3D;printer;convert;gcode;stl;obj;amf;SLA
> StartupNotify=false
> StartupWMClass=orca-slicer
> X-Desktop-File-Install-Version=0.28
> X-Flatpak=io.github.softfever.OrcaSlicer
> ```
its not work for me :/ same behavior like before(https://github.com/OrcaSlicer/OrcaSlicer/issues/11541), but additional now the home tab also not work.

itay-grudev, 2025-12-29:
As a temporary workaround and a correction to the previously proposed desktop file, keep in mind that the Flatpak doesn't have permissions to access `/usr/share/glvnd/...`. So you should copy the file to your home directory like so:

```bash
mkdir -p ~/.local/share/glvnd/egl_vendor.d                                     
cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/.local/share/glvnd/egl_vendor.d
```

Then your shortcut file in `~/.local/share/applications/io.github.softfever.OrcaSlicer.desktop` should look like this:

```
[Desktop Entry]
Name=OrcaSlicer
GenericName=3D Printing Software
Icon=io.github.softfever.OrcaSlicer
Exec=/bin/bash -c '__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=$HOME/.local/share/glvnd/egl_vendor.d/>
Terminal=false
Type=Application
MimeType=model/stl;model/3mf;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;x-scheme-ha>
Categories=Graphics;3DGraphics;Engineering;
Keywords=3D;Printing;Slicer;slice;3D;printer;convert;gcode;stl;obj;amf;SLA
StartupNotify=false
StartupWMClass=orca-slicer
X-Desktop-File-Install-Version=0.28
X-Flatpak=io.github.softfever.OrcaSlicer
```

irexiz, 2026-03-27:
> Alright, so I think I figured this out ([#12308](https://github.com/OrcaSlicer/OrcaSlicer/pull/12308)):
> 
> This problem occurs whenever GLEW and wxWidgets do not agree to both either use EGL or GLX. This problem is compounded by some distributions un-vendoring wxWidgets in their builds, which is e.g. what NixOS is doing. When you build wxWidget with `--disable-glcanvasegl` from nixpgs, patchting it like this:
> 
>   wxGTK' =
>     (wxGTK31.override {
>       withCurl = true;
>       withPrivateFonts = true;
>       withWebKit = true;
>     }).overrideAttrs
>       (old: {
>         configureFlags = old.configureFlags ++ [
>           # Disable noisy debug dialogs
>           "--enable-debug=no"
>          # Disable EGL to work around bug 6433
>           "--disable-glcanvasegl"
>         ];
>       });
> 
> orca-slicer works perfectly.

Confirming that this worked on NixOS, Wayland, Nvidia 595.45.04

daef, 2026-11-03:
no stale bot, bad stale bot... *activity*