OrcaSlicer/OrcaSlicer #6434: Wen importing a multi-plate 3mf that was saved with BambuStudio objects on plates are not centered.

sonjamichelle, 2024-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1 and lower

### Operating System (OS)

Windows

### OS Version

windows 11 (latest updates as of 8-12-24)

### Additional system information

7th Gen Intel i7
64GB RAM
NVIDIA RTX 3900

### Printer

BambuLabs X1C

### How to reproduce

Drag or load an 3mf print profile that had previously been saved with BambuStudio.

### Actual results

All objects are off center. Instead of 128x128 they are placed at 133x135 or there abouts. Each subsequent plate beyond Plate 04 sees the offset higher on the pplate.

### Expected results

All objects should be placed as close to the center of each plate or near the orginal placement as deemed by the print profile.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16580533/log.zip)
[M-3+Predator+Prop.zip](https://github.com/user-attachments/files/16580538/M-3%2BPredator%2BProp.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

sharky98, 2024-08-14:
Maybe related to #3641 and #3852?

closed 2024-11-20 by github-actions[bot]

sonjamichelle, 2024-11-20:
So, it is an issue that has been reported by many and collaborated by many and will be marked as a "non-issue" with no resolution planned.

Is that that the official stance?

sonjamichelle, 2024-11-20:
Here is a possible solution. 

When you try to upload a print profile created by OrcaSlicer to MakerWorld you get told that the profile was not created by BambuStudio. So at least BambuStudio is identifying itself in some form or other.

USE that identification to choose what plate spacing to use.

sonjamichelle, 2024-11-20:
NOW I am seeing that even though I created the 3MF WITH OrcaSlicer, when I load multiple plates 4 and beyond NO LONGER HAVE THE PARTS IN THE CORRECT LOCATION!

So, closing this issue is NOT warranted!