OrcaSlicer/OrcaSlicer #6442: Document 3rd party connections

coreyward, 2024-08-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Orca Slicer reaches out to various IP addresses and websites over the network on ports 80, 8000, 8080, and 10001 (possibly others). There are a handful of clear requests to Bambu or AWS ELB servers, but the nature of most of the remaining requests is unclear. Many do not have any DPI-indicated hostnames associated, and both the data being transferred as well as who it is being sent to are unclear. For example, I have observed requests to connect to `chats.partialqyrlkfa.com`, `64.42.179.82`, `173.234.30.210`, `118.107.244.104`, and many others.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Is it possible to get these connections documented, including the nature of the information being sent and the purpose it is being sent for? It would also be helpful to know which are required for the software to operate (or which functions depend on each connection).

### Describe alternatives you've considered

Using other slicers that don't send data to unidentified third parties unannounced.

gleep52, 2024-09-03:
I am currently attempting to figure out why my print jobs seem to send to BambuLabs' servers fine, but then take minutes to "download" them to the printer.  Looking at my firewall I am seeing a lot of this same strange traffic associated with my printer.  I have not yet tested this same issue with the official Bambu Studio app - but am curious if it's the servers at BambuLabs that use this strange domain chats.partialqyrlkfa.com and the other misc IPs?

coreyward, 2024-12-04:
Sure is stale because no maintainers have chimed in. 👀

coreyward, 2025-03-06:
Not stale, just ignored. Almost like @SoftFever has something to hide…

SoftFever, 2025-08-13:
Can you provide more information?  
Orca will only ping GitHub to check for updates.  

If you use Bambu, their closed-source plugin will connect to their servers.  
However, I do not believe it will connect to chats.partialqyrlkfa.com such address or similar websites. (Although I can't say it 100% sure as it's closed source) 

Could you please double-check if you have configured any of your printer's host names or Device UI to such an address?
[image]

pieterdevriesch, 2025-08-13:
I agree that it's the Bambu plugin, Bambu Studio gives exactly the same activity, and in both it only happens when you try to start a video stream. Which still works despite the blocked connections funnily enough. 

So I guess this issue can be closed since it's Bambu's issue, there's an issue on their git about this already now. Thanks for the update.

closed 2025-08-13 by SoftFever