OrcaSlicer/OrcaSlicer #6445: Add used filament in UI in layer slider

przemowcf, 2024-08-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am missing information about used filament (total) at currently viewed layer.

Knowing this I can plan moment to change the spool if print going to be a long one (and use more filament than I have on spool). And I could then add at specific `layer_change` macro to change filament (like `M600`)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'm thinking adding this at the layer preview slider:

*this screenshot also mention the gcode - this matter I'll add as different issue/feature-request*
[image]

### Describe alternatives you've considered

I'd be happy if it would be already as comment in gcode preview displayed (like in screenshot).  ➡️  feature request: #6444  .

kylek29, 2024-08-15:
I do think that would be useful. That slider section could use a tad more info and a revisit ever since they added per object printing .. it'd also be nice to have the object's relative layer (e.g. 1/56 for OBJB, 302/1024 for Job).

closed 2024-11-21 by github-actions[bot]

digitalresistor, 2025-07-24:
Anyway to re-open this feature request without re-creating the feature request entirely? This would be great to get a better idea for how much filament will be used by what time in the print, so that I can estimate when a spool is about to run out and at about what time into the print.

przemowcf, 2025-07-24:
@digitalresistor I think just create new Issue and ref this issue 6445 and 6444 .
As worlk-around I written a small script in python that process the gcode and sum up all E axis movement, and printout the length at each layer ... Here is the script: https://gist.github.com/przemowcf/8ad68e35d4caadba96f94e7b3621f990