OrcaSlicer/OrcaSlicer #6448: Makes no sense to make >100% overhangs use the same speed as bridging speed

ckgdmm1, 2024-08-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I find it strange that >100% overhang speeds are printed using bridge speeds and there's no way to specify its own speed, noticed when printing overhang tests. Bridge speeds often can be faster because of end point anchors, infact the default settings have the speeds higher, but if you want to print something like the image, it doesn't make sense for the nozzle to slow down and then speed up for the steepest part of the overhang

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice and probably very easy to separate the speed settings for >100% overhang and bridges

### Additional context

[image]

tome9111991, 2024-08-13:
Hi.
Check:

https://github.com/SoftFever/OrcaSlicer/issues/5788#issuecomment-2182741408

freakydude, 2024-09-14:
@tome9111991 

Seems to be more or less fixed (intentionally) with the "slow down for curled perimeters" turned off. 

But it still feels like a compromise...

My concept would be
introduce a new additional overhang speed option -> 99.5-100% or something like that. And use that for setting "real" bridge speeds. 
Maybe it could be a complete replacement of bridge speed / bridge fan speed.

slow down on
[image]

slow down off
[image]

staal54a, 2024-12-16:
Bumping since overhang/bridge overlap is still an issue

Rimjakk, 2024-12-16:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

Wouldn't be stale if it wasn't just the same issue enduring forever without any fix or notice of a fix. 
I'd be happy if we could just "paint" bridging like we can "paint" supports.  

[image]