OrcaSlicer/OrcaSlicer #6449: **`Open Containing Folder`** opens the actual file instead of the folder

discip, 2024-08-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

https://github.com/SoftFever/OrcaSlicer/commit/7082e945b12f2e37584c3aa1dc68f513c8b4eee0

### Operating System (OS)

Linux

### OS Version

Manjaro Linux

### Additional system information

# System Details Report
---

## Report details
- **Date generated:**                              2024-08-13 15:47:20

## Hardware Information:
- **Hardware Model:**                              ASUS ROG STRIX X670E-E GAMING WIFI
- **Memory:**                                      32,0 GiB
- **Processor:**                                   AMD Ryzen™ 9 7950X3D × 32
- **Graphics:**                                    AMD Radeon™ RX 7900 XTX
- **Disk Capacity:**                               2,0 TB

## Software Information:
- **Firmware Version:**                            2007
- **OS Name:**                                     Manjaro Linux
- **OS Build:**                                    rolling
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.8.12-3-MANJARO

### Printer

-

### How to reproduce

1. Launch Orca and navigate to the Home screen,
2. right click on any project file and
3. click **`Open Containing Folder`**

### Actual results

Instead of the containing folder, the actual file is opened in another instance of **Orca**.

### Expected results

The folder containing the project file should open.

### Project file & Debug log uploads

Not related to a specific project file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2024-08-13:
@Noisyfox
There seem to be even more Linux specific quirks.
Any idea on that one?

Noisyfox, 2024-08-14:
Yeah, linux does not have a common "open containing folder" command. I think the best option is to open the folder without the file been seleceted automaically.

discip, 2024-08-14:
So, ... is there anything that could be done to solve this?

Noisyfox, 2024-08-14:
> I think the best option is to open the folder without the file been seleceted automaically, instead of using a disto-specific command.

This.

discip, 2024-08-14:
Sorry, I'm not able to follow.😅
What exactly do you mean?

discip, 2024-08-14:
> So, instead, for Linux I think it's much easier to just do step 1 without step 2,

I totally agree, but I don't know how to achieve the following:

> which should be achieved by a `std::filesystem::path::parent_path` then an `xdg-open` command.

PR #6469 (2024-08-18): Fix "Open Containing Folder" on Linux
This fixes #6449.

Detailed descussions are in the original issue.
closed 2024-08-18 by SoftFever