OrcaSlicer/OrcaSlicer #6451: Missing RFID support for PETG-CF

oernster, 2024-08-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have been reliably informed (I can't test this as I don't have said filament) but they don't want to raise a PR, that PETG CF RFID tags that are supported on Bambu Studio are not supported on Orca Slicer. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like PETG-CF RFID Bambu spool tags to be recognised by Orca Slicer, please.

### Describe alternatives you've considered

N/A

Choku81, 2024-08-19:
Bambu Studio doesn't even support PETG HF, I've only gotten PETG HF to work on the mobile app.

I'd like to upvote this as Printing with PETG HF seems tough right now.

inhumantsar, 2024-08-23:
@Choku81 i don't have it installed but bambu studio apparently does support it. comes down in a configuration update that's offered at startup.

thankfully a kind soul in this [thread](https://forum.bambulab.com/t/petg-hf-print-profiles/88834/11) diffed the generic PETG and official PETG-HF profiles in case anyone wants to create a custom profile for now:

[image]

closed 2024-11-30 by github-actions[bot]